OSSystems/meta-browser #242: chromium-ozone-wayland: segfaults on pull-up menus

MastaG, 2019-03-26:
Hi I'm succesfull running chromium-ozone-wayland on gnome 3.32.
It works very fast and accelerated video decoding using v4l2 is working too which is a big plus.

Only thing is that it always seems to segfault when selecting items from pull-up click menu's.
So for example you right click.. some options appear such as copy, paste, etc..
Now when you click one, for example copy.. then it freezes and segfaults.

The same happens when you click the 3-dot button the upper-right corner and select anything from the list, like settings for example.

Also when I close the chromium window the same segfault occurs.

Other than that, it runs great.

So I launch it using these options:
>>--use-gl=egl --enable-gpu-rasterization --enable-impl-side-painting \
>>--disable-features=VizDisplayCompositor --ozone-platform=wayland --in-process-gpu \
>> --ignore-gpu-blacklist

And the following appears on the console:
>>[7859:7859:0326/202045.194349:ERROR:desktop_window_tree_host_platform.cc(445)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::InitModalType(ui::ModalType)
[7859:7859:0326/202045.194534:ERROR:desktop_window_tree_host_platform.cc(455)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsAnimatingClosed() const
[7859:7859:0326/202045.215090:ERROR:desktop_window_tree_host_platform.cc(468)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SizeConstraintsChanged()
[7859:7859:0326/202045.417499:ERROR:desktop_window_tree_host_platform.cc(439)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SetWindowIcons(const gfx::ImageSkia &, const gfx::ImageSkia &)
[7859:7859:0326/202045.417660:ERROR:desktop_window_tree_host_platform.cc(371)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::SetWindowTitle(const base::string16 &)
[7859:7859:0326/202045.511890:ERROR:desktop_window_tree_host_platform.cc(309)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Activate()
[7859:7859:0326/202045.734647:ERROR:desktop_window_tree_host_platform.cc(226)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsVisible() const
[7859:7898:0326/202046.222854:ERROR:buffer_manager.cc(488)] [.BrowserCompositor-0x9fa2000]GL ERROR :GL_INVALID_FRAMEBUFFER_OPERATION : glBufferData: <- error from previous GL command

Everything works great, but once I select an item from some pull-up menu, I get:
>>[7859:7859:0326/202047.896006:ERROR:desktop_window_tree_host_platform.cc(377)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::ClearNativeFocus()
[7859:7859:0326/202056.591699:ERROR:desktop_window_tree_host_platform.cc(267)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::GetWindowPlacement(gfx::Rect *, ui::WindowShowState *) const
Segmentatiefout (geheugendump gemaakt)

Any ideas?

msisov, 2019-03-26:
The Wayland support is still under development and is not officially shipped. 

The problem you have seem to be related to the gpu operations (buffer creation/handling), which results to corrupted frame buffer. 

You’d need to provide us with a backtrace and submit the problem to crbug.com

closed 2019-03-26 by msisov

MastaG, 2019-03-26:
Alright my friend :)
I'll do my best to create one and submit it to crbug.com
Which are the required gn flags to properly generate a backtrace?
I guess something like symbol_level=2 ?

msisov, 2019-03-26:
should be enough