OSSystems/meta-browser #246: Chromium 73 bitbake problem

peel70, 2019-04-02:
Hello,
I've a problem with YOCTO's Thud branch.
I've downloaded OSSystems's master branch and try to compile chromium-x11 but I've the following error, at the end of the process.

| FAILED: chrome
| python "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- x86_64-poky-linux-g++  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/developer/yocto/thud/build/intel/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.75-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,now -Wl,-z,relro -Wl,-z,defs -Wl,--as-needed -m64 -Wl,-O2 -Wl,--gc-sections -rdynamic -pie -Wl,-rpath-link=. -Wl,--disable-new-dtags -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group   -latomic -ldl -lpthread -lrt -lX11 -lX11-xcb -lxcb -lXcomposite -lXcursor -lXdamage -lXext -lXfixes -lXi -lXrender -lXtst -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -lxml2 -ldbus-1 -lXss -lwebpdemux -lwebpmux -lwebp -lfreetype -ljpeg -lexpat -luuid -lXrandr -lresolv -lgio-2.0 -lasound -lm -lz -lpci -lpangocairo-1.0 -lpango-1.0 -lcairo -latk-1.0 -latk-bridge-2.0 -latspi -lFLAC -lgtk-3 -lgdk-3 -lcairo-gobject -lgdk_pixbuf-2.0 -lxslt
| /home/developer/yocto/thud/build/intel/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.75-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/../../libexec/x86_64-poky-linux/gcc/x86_64-poky-linux/8.2.0/ld: obj/third_party/webrtc/modules/audio_coding/libisac_c.a: error adding symbols: malformed archive
| collect2: error: ld returned 1 exit status
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/developer/yocto/thud/build/intel/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.75-r0/temp/log.do_compile.26731)
ERROR: Task (/home/developer/yocto/thud/poky/meta-browser/recipes-browser/chromium/chromium-x11_73.0.3683.75.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9092 tasks of which 10 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/developer/yocto/thud/poky/meta-browser/recipes-browser/chromium/chromium-x11_73.0.3683.75.bb:do_compile
Summary: There were 5 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Please let me know.
Regards
Enrico

msisov, 2019-04-02:
https://github.com/OSSystems/meta-browser/issues/194#issuecomment-440411509

peel70, 2019-04-02:
thankyou msisov!
it work!
best regards
Enrico

closed 2019-04-02 by peel70