OSSystems/meta-browser #249: chromium-x11 fails on actual master

nandra, 2019-04-04:
I'm trying to build chromium-x11 (master branch) other layers have thud release and get following:
```
FAILED: yocto_native/obj/base/base/task_annotator.o
g++  -MMD -MF yocto_native/obj/base/base/task_annotator.o.d -DUSE_SYMBOLIZE -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DBASE_IMPLEMENTATION -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_32 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_26 -I../.. -Iyocto_native/gen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -m64 -march=x86-64 -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -fno-omit-frame-pointer -g0 -fvisibility=hidden -I/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/glib-2.0 -I/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib/glib-2.0/include -I/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include -O2 -fno-ident -fdata-sections -ffunction-sections -std=gnu++14 -Wno-narrowing -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot-native/usr/include -isystem/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot-native/usr/include -O2 -pipe -c ../../base/debug/task_annotator.cc -o yocto_native/obj/base/base/task_annotator.o
In file included from ../../base/trace_event/trace_event.h:25:0,
                 from ../../base/debug/task_annotator.cc:14:
../../base/trace_event/trace_arguments.h:486:20: error: enclosing class of constexpr non-static member function �~@~Xsize_t base::trace_event::StringStorage::size() const�~@~Y is not a literal type
   constexpr size_t size() const { return data_ ? data_->size : 0u; }
```

Anybody facing same issue? Thanks.

rakuco, 2019-04-04:
That part of the build's happening on the native side, what GCC version does your system have?

nandra, 2019-04-04:
Ubuntu 16.04:
```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.11' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11) 
```
should we bump gcc to 7.x or so? Thanks.

rakuco, 2019-04-04:
Yes, GCC 5 is very old by Chromium standards and doesn't implement all the C++11/C++14 features we need, so it's always a good idea to use the latest GCC version you can get.

nandra, 2019-04-04:
GCC 5 have c++11 (not sure about c++14)> I'l try to bump to GCC 7 and re-test. Thanks. I'll report here. :+1:

nandra, 2019-04-05:
@rakuco ok bumped gcc to 7.4 and it almost compile (break at 99% :( ) with following:
```
[37642/37645] rm -f obj/chrome/browser/ui/libui.a && "x86_64-poky-linux-ar" -T -r -c -s -D obj/chrome/browser/ui/libui.a @"obj/chrome/browser/ui/libui.a.rsp"
[37643/37645] rm -f obj/chrome/browser/libbrowser.a && "x86_64-poky-linux-ar" -T -r -c -s -D obj/chrome/browser/libbrowser.a @"obj/chrome/browser/libbrowser.a.rsp"
[37644/37645] touch obj/chrome/browser_dependencies.stamp
[37645/37645] python "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- x86_64-poky-linux-g++  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,now -Wl,-z,relro -Wl,-z,defs -Wl,--as-needed -m64 -Wl,-O2 -Wl,--gc-sections -rdynamic -pie -Wl,-rpath-link=. -Wl,--disable-new-dtags -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group   -latomic -ldl -lpthread -lrt -lX11 -lX11-xcb -lxcb -lXcomposite -lXcursor -lXdamage -lXext -lXfixes -lXi -lXrender -lXtst -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -lxml2 -ldbus-1 -lXss -lwebpdemux -lwebpmux -lwebp -lfreetype -ljpeg -lexpat -luuid -lXrandr -lresolv -lgio-2.0 -lasound -lm -lz -lpci -lpangocairo-1.0 -lpango-1.0 -lcairo -latk-1.0 -latk-bridge-2.0 -latspi -lFLAC -lgtk-3 -lgdk-3 -lcairo-gobject -lgdk_pixbuf-2.0 -lxslt
FAILED: chrome
python "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- x86_64-poky-linux-g++  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,now -Wl,-z,relro -Wl,-z,defs -Wl,--as-needed -m64 -Wl,-O2 -Wl,--gc-sections -rdynamic -pie -Wl,-rpath-link=. -Wl,--disable-new-dtags -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group   -latomic -ldl -lpthread -lrt -lX11 -lX11-xcb -lxcb -lXcomposite -lXcursor -lXdamage -lXext -lXfixes -lXi -lXrender -lXtst -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -lxml2 -ldbus-1 -lXss -lwebpdemux -lwebpmux -lwebp -lfreetype -ljpeg -lexpat -luuid -lXrandr -lresolv -lgio-2.0 -lasound -lm -lz -lpci -lpangocairo-1.0 -lpango-1.0 -lcairo -latk-1.0 -latk-bridge-2.0 -latspi -lFLAC -lgtk-3 -lgdk-3 -lcairo-gobject -lgdk_pixbuf-2.0 -lxslt
/home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/../../libexec/x86_64-poky-linux/gcc/x86_64-poky-linux/8.2.0/ld: obj/third_party/webrtc/modules/audio_coding/libisac_c.a: error adding symbols: malformed archive
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
```

ideas? Thx.

rakuco, 2019-04-05:
> /home/builder/test/.build-x86/tmp/work/corei7-64-poky-linux/chromium-x11/73.0.3683.86-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/../../libexec/x86_64-poky-linux/gcc/x86_64-poky-linux/8.2.0/ld: obj/third_party/webrtc/modules/audio_coding/libisac_c.a: error adding symbols: malformed archive

That's https://github.com/OSSystems/meta-browser/issues/194#issuecomment-440411509

rakuco, 2019-04-11:
Did it work? Can we close this issue?

nandra, 2019-04-11:
Let me give couple more days I'll test and let you know. Thanks.

nandra, 2019-04-14:
With #194 comment works perfectly fine. Thanks.

nandra, 2019-04-14:
Maybe it would be good to put this info to README so other who hit same issue are aware of solution ;). Thx.

closed 2019-04-14 by nandra