OSSystems/meta-browser #251: chromium-ozone-wayland-73.0.3683.103-r0 do_compile failed

kraj, 2019-04-17:
I am seeing below failures on qemux86-64 using gcc as default compiler. Its using master of oe-core.

https://errors.yoctoproject.org/Errors/Details/237664/

kraj, 2019-04-17:
@msisov @rakuco let me know if you need more info than above.

msisov, 2019-04-17:
@kraj it's the wayland scanner problem. What is your env?

rakuco, 2019-04-17:
I'm pretty sure this is #241

kraj, 2019-04-17:
> @kraj it's the wayland scanner problem. What is your env?

to be honest I did not look but it sounded in walyand scanner python scriptish to me.

petegriffin, 2019-04-25:
I just looked at @kraj log and it is indeed the wayland scanner issue. Temp workaround is to install wayland-scanner on your build host.

However after that workaround I get further compile errors with gcc-8.2, see log here https://pastebin.linaro.org/view/f14a2ef0

rakuco, 2019-05-02:
Closing as a duplicate of #241.

closed 2019-05-02 by rakuco