OSSystems/meta-browser #253: Segfault when launching chromium with ozone-platform=wayland

MastaG, 2019-04-24:
Hi,

I'm building chromium with:
use_ozone=true
ozone_auto_platforms=false
ozone_platform_headless=true
ozone_platform_wayland=true
ozone_platform_x11=false
use_system_libwayland=true
use_system_minigbm=false
use_system_libdrm=false
use_exynos_minigbm=true

Note that  I'm using the included libdrm and exynos minigbm because ARM's libgbm is too outdated.

I'm running Chromium with:
--use-gl=egl
--enable-gpu-rasterization
--enable-impl-side-painting
--ozone-platform=wayland
--in-process-gpu
--ignore-gpu-blacklist

Note that the '--in-process-gpu' flag used to be required or else Chromium would segfault on launch.

But after this commit: https://github.com/OSSystems/meta-browser/commit/164e929d5dc875dc6523e54dc9ebf58e8a31e720
Chromium always segfaults, even when passing the --in-process-gpu flag.

Wayland debug:
[wayland_debug.txt](https://github.com/OSSystems/meta-browser/files/3111338/wayland_debug.txt)

Gdb trace:
Thread 1 "chromium-bin" received signal SIGSEGV, Segmentation fault.
0x017c458e in make_unique<ui::WaylandBufferManager::Surface, ui::WaylandWindow*&, ui::WaylandConnection* const&> () at /usr/include/c++/v1/memory:3132
3132	in /usr/include/c++/v1/memory

msisov, 2019-04-29:
@MastaG, according to the wayland_debug, you don't have the zwp_linux_dmabuf protocol available. 

PS I will fix this by falling back to either software mode or by falling back to using wl_egl_surface (depending whether in-process-gpu is passed or not).

msisov, 2019-04-29:
can you paste full bt?

MastaG, 2019-04-29:
Hi @msisov 
Thanks for looking into this.
I'm using the ARM mali-T628 r12p004 drivers for the odroid xu3.
They're only distributed as binaries and very outdated.
It provides it's own version of libEGL and libwayland-egl, so it's quite possible it misses zwp_linux_dmabuf support.

So if you're going to create a workaround for that, then please don't as there's a open-source mesa driver (panfrost) currently in development which is progressing really well.

I was fighting a bit with gdb as it wouldn't cooperate with me, but I'll try to make a full bt when I get home and post it here.

MastaG, 2019-05-02:
Well I'm not so experienced with gdb.
I'm running it with:
gdb --args /usr/lib/chromium/chromium-bin --use-gl=egl --enable-gpu-rasterization --enable-impl-side-painting --ozone-platform=wayland --in-process-gpu --ignore-gpu-blacklist

Here's the full backtrace on all threads, but it's the first one that segfaults.
[bt_all_threads.txt](https://github.com/OSSystems/meta-browser/files/3137264/bt_all_threads.txt)

msisov, 2019-05-06:
I guess we have discussed it in the IRC, right? The problem was with the zwp linux dmabuf not being available. If so, closing it.

closed 2019-05-06 by msisov