OSSystems/meta-browser #258: arm64 GCC failures

msisov, 2019-05-10:
It turned out the arm64 builds fail with gcc :

> include/arm_acle.h:65:1: error: inlining failed in call to always_inline 'uint32_t __crc32cd(uint32_t, uint64_t)': target specific option mismatch
>  __crc32cd (uint32_t __a, uint64_t __b)
>  ^~~~~~~~~
> ../../third_party/crc32c/src/src/crc32c_arm64.cc:94:21: note: called from here
>      crc ^= __crc32cd(0, t0);
>             ~~~~~~~~~^~~~~~~
>

kraj, 2019-05-10:
this seems more like a gcc problem to me. which version of gcc is showing this issue ?

msisov, 2019-05-10:
Both gcc 7.3 and gcc 8

kraj, 2019-05-10:
gcc9 is on its way in ( currently in master-next ), I will be interested to know if its failing with gcc9 as well. then we can see if we can get upstream gcc's attention on it.

msisov, 2019-05-13:
The problems seems to be in this recent patch - https://chromium-review.googlesource.com/c/chromium/src/+/1492489.  

It used to be setting armv8-a+crc+crypto CFLAG for gcc, but doesn't do so now. Adding else condition with cflags = [ "-march=armv8-a+crc+crypto" ] fixes the issue.

@rakuco is it right thing to do? if I set the march to be armv8 and conditionally add crc and crypto, it doesn't understand those.

rakuco, 2019-05-13:
Good catch. https://bugs.chromium.org/p/chromium/issues/detail?id=934016#c4 mentions CrOS setting `-march`, and I think we do something similar (though I don't 100% recall what we've changed in `oe-clang-fixes.patch`). If everything builds fine with an `else` I'd say go for it.

msisov, 2019-05-13:
@rakuco, oh. dape has already fixed it the same way. I just need to backport the patch and that's it.

msisov, 2019-05-13:
fixed.

closed 2019-05-13 by msisov