OSSystems/meta-browser #265: Road to firefox 68

ashie, 2019-05-29:
According to [Firefox release Calendar](https://wiki.mozilla.org/Release_Management/Calendar), next ESR of Firefox will be shipped on 2019-07-09. So I'm now trying to port alpha or beta versions of it to Yocto.

  * https://github.com/webdino/meta-browser/tree/firefox-68-wip

Current status:

* [An alpha version already get working on AArch64](https://github.com/webdino/gecko-embedded/issues/70#issuecomment-482408084) although some additional host tools and hacks are required.
* Beta version isn't succeeded to compile yet since Rust-1.34 is required (current meta-rust is Rust-1.33).
* Most backport patches are removed since they are already merged into upstream.

Issues:

* Rust-1.34 is required so we need to update meta-rust.
* Node.js 8.11 or later is required as a native toolchain.
  * It's already resolved. The master branch of meta-openembedded satisfies this condition.
* [cbindgen](https://github.com/eqrion/cbindgen) is required but probably no layer provides it. So we need create a new recipe.
  * I'm now using host's one.
* autoconf2.13 is required as a host tool.
  * I'll add a patch to omit it as same as current master of meta-browser.
* Recent FIrefox cannot build against GCC, so meta-clang will be needed.
  * I compiled the alpha version with it.
  * I'm not sure how to switch to it properly. I'm now adding the following lines to my local.conf:
```
TOOLCHAIN_pn-firefox = "clang"
AS_pn-firefox="${TARGET_PREFIX}clang"
```

otavio, 2019-05-29:
Thank you for your hard work; looking forward to it :1st_place_medal:

ashie, 2019-05-30:
>     * [cbindgen](https://github.com/eqrion/cbindgen) is required but probably no layer provides it. So we need to create a new recipe.

Now I've noticed that creating the recipe is very easy.
According to the [README.md of meta-rus](https://github.com/meta-rust/meta-rust#building-a-rust-package), we can generate it automatically by [cargo-bitbake](https://crates.io/crates/cargo-bitbake).

```
$ cargo install cargo-bitbake
$ git clone https://github.com/eqrion/cbindgen.git
$ cd cbindgen
$ cargo bitbake
```

Now I got a template "cbindgen_0.8.7.bb".

In addition, at least following two modification are required:

* Set appropriate `LIC_FILES_CHKSUM`
* Add `BBCLASSEXTEND = "native"` to use as a native toolchain

I temporarily [added the recipe to my branch](https://github.com/webdino/meta-browser/blob/firefox-68-wip/recipes-browser/firefox/cbindgen_0.8.7.bb).

The remaining issue for cbindgen is who should have this recipe? meta-rust? or meta-browser?

ashie, 2019-05-30:
> The remaining issue for cbindgen is who should have this recipe? meta-rust? meta-browser? or any other layer?

It seems that all prominent users of cbindgen are Mozilla related projects, so I think having in meta-browser is better than others for a while.

ashie, 2019-06-06:
>     * Rust-1.34 is required so we need to update meta-rust.

https://github.com/meta-rust/meta-rust/pull/241

ashie, 2019-07-03:
>     * [ ]  autoconf2.13 is required as a host tool.
>       * I'll add a patch to omit it as same as current master of meta-browser.

Instead of adding a patch to avoid running autoconf2.13, I'll add a new recipe to introduce the native autoconf2.13 command, because adding a patch makes it bothersome to update.
I think adding the recipe for autoconf2.13 to this layer is reasonable since there is no other user who uses such a too old version of autoconf in these days.

https://github.com/webdino/meta-browser/tree/firefox-68-wip/recipes-devtools/autoconf2.13

otavio, 2019-07-03:
It all depends how big is the patch. Looking at your recipe for the `autoconf2.13` it seems is it lacking a perl native inherit, no?

If small, I'd prefer to avoid the old autoconf as it has some intricate relationship with the host.

ashie, 2019-07-04:
> I'd prefer to avoid the old autoconf as it has some intricate relationship with the host.

Hmm. I also don't want to introduce such an obsolete technology.
OK, I'll consider another way.

The problem of the current way is creating a patch for a pre-generated configure manually.
It's bothersome and not reproducible, no one knows how to create it except me.
I'll try to make a script to create it.

otavio, 2019-07-04:
> The problem of the current way is creating a patch for a pre-generated configure manually.
> It's bothersome and not reproducible, no one knows how to create it except me.
> I'll try to make a script to create it.

Adjusting the `configure.ac` file and regenerating it during build, does not help?

Also, a patch upstream fixing this would solve both problems.

PR #284 (2019-07-16): Update to firefox 68.0esr
* Update to firefox 68.0esr
* Support only Clang via meta-çlang, you can't use GCC to build it
    * The dependency isn't added to conf/layer.conf yet because:
        * Chromium still uses GCC
        * meta-clang isn't required if you don't build firefox
* Add a new recipe for a toolchain "cbindgen" which is required to build modern firefox
* Update Debian's patch set
* Remove or update obsolete patches
* firefox-l10n is also updated
* Update README to add more descriptions about the recipe for firefox

Fix #265
closed 2019-07-16 by otavio