OSSystems/meta-browser #272: Failed to run Chromium under X11 with Raspberry Pi 3 machine

fbertux, 2019-06-17:
Hi all!

I'm trying to run Chromium on raspberrypi3 machine. Currently I can open Chromium Browser, but when I click in something browser suddenly stops and show error:

```shell
root@host:~# chromium --no-sandbox

(process:583): Gtk-WARNING **: 15:38:36.414: Locale not supported by C library.
        Using the fallback 'C' locale.
[583:615:0617/153836.775677:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
pure virtual method called
terminate called without an active exception
Aborted
```

I'm using meta-browser on latest master revision, oe-core and meta-raspberrypi layers on warrior branch. Chromium version is 75.0.3770.90 and Linux version 4.14.112. The distro features opengl and x11 are enabled and I'm using vc4graphics machine feature.

Does anyone is running Chromium successfully with raspberrypi3 machine, or I'm missing something?

Thanks!

otavio, 2019-06-17:
Any idea folks? We tried both vc4 and userland driver and both behave the same.

rakuco, 2019-06-17:
I don't have a RPi myself to test this, would it be possible to check if it's reproducible on IA?

kraj, 2019-06-17:
this seems to be related to or coming from d-bus AFAICT, it is unrelated to vc4graphics. but if you wan to use vc4graphics then I would recommend using kernel 4.19 for better results.

otavio, 2019-06-19:
@rakuco 
> I don't have a RPi myself to test this, would it be possible to check if it's reproducible on IA?

It should be; we had it with `vc4graphics` and `userland` so it is not driver specific. Most likely a missing dependency or a bug.

The image we used to test was a very minimal image, it had just Chromium and x11-base. Another important aspect is that we launch it from SSH or serial console, so not inside the X11 existing session.

@kraj 
> this seems to be related to or coming from d-bus AFAICT, it is unrelated to vc4graphics. but if you wan to use vc4graphics then I would recommend using kernel 4.19 for better results.

We ended moving to QtWebEngine for this project. That said, it does not seem to be d-bus related as the web page opens fine, it dies when we click on a link.

msisov, 2019-06-21:
I have an rpi3 and see the same issue. I will look into it

msisov, 2019-06-21:
It looks like it is this issue - https://bugs.chromium.org/p/chromium/issues/detail?id=971362
and fixed by - https://chromium-review.googlesource.com/c/chromium/src/+/1645366

Verifying it.

otavio, 2019-06-21:
Fixed by #276

closed 2019-06-21 by otavio

rakuco, 2019-06-21:
@msisov thanks for taking care of this!