OSSystems/meta-browser #274: chromium: Update to 75.0.3770.100

rakuco, 2019-06-19:
Release announcement: https://chromereleases.googleblog.com/2019/06/stable-channel-update-for-desktop_18.html

Targets and platforms we need to check:

#### chromium-x11:
- [ ] rocko, GCC 7, MACHINE="qemuarm64"
- [ ] sumo, GCC 7, MACHINE="raspberrypi3"
- [ ] thud, GCC 8, MACHINE="intel-core-i7-64"
- [ ] thud, GCC 8, MACHINE="intel-core2-32"

#### chromium-ozone-wayland:
- [ ] sumo, GCC 7, MACHINE="raspberrypi3"
- [ ] sumo, GCC 7, MACHINE="beaglebone"
- [ ] thud, GCC 8, MACHINE="m3ulcb"
- [ ] thud, clang, MACHINE="qemuarm64"

rakuco, 2019-06-19:
I spoke to @msisov on IRC and he's unable to test m3ulcb today. Given this is a small update and everything else passed, I'm going to proceed and send a PR.

otavio, 2019-06-19:
As it is a minor update, I'd be fine in merging it.

closed 2019-06-19 by rakuco

msisov, 2019-06-21:
thud, GCC 8, MACHINE="m3ulcb" also passed

PR #275 (2019-06-19): chromium: Update to 75.0.3770.100.
Release announcement:
https://chromereleases.googleblog.com/2019/06/stable-channel-update-for-desktop_18.html

Test-built with
---------------
* chromium-x11
  * rocko, GCC 7, MACHINE="qemuarm64"
  * sumo, GCC 7, MACHINE="raspberrypi3"
  * thud, GCC 8, MACHINE="intel-core-i7-64" and MACHINE="intel-core2-32"
* chromium-ozone-wayland:
  * sumo, GCC 7, MACHINE="raspberrypi3"
  * sumo, GCC 7, MACHINE="beaglebone"
  * thud, clang, MACHINE="qemuarm64"

Fixes #274.
commit e883c129 (2019-06-19): chromium: Update to 75.0.3770.100.

Release announcement:
https://chromereleases.googleblog.com/2019/06/stable-channel-update-for-desktop_18.html

Test-built with
---------------
* chromium-x11
  * rocko, GCC 7, MACHINE="qemuarm64"
  * sumo, GCC 7, MACHINE="raspberrypi3"
  * thud, GCC 8, MACHINE="intel-core-i7-64" and MACHINE="intel-core2-32"
* chromium-ozone-wayland:
  * sumo, GCC 7, MACHINE="raspberrypi3"
  * sumo, GCC 7, MACHINE="beaglebone"
  * thud, clang, MACHINE="qemuarm64"

Fixes #274.