OSSystems/meta-browser #277: [chromium-x11] Phyton compilation errors

denis-shienkov, 2019-06-24:
I have an yocto 'pyro' layer, where are I use the chromium v52. It was compiled fine... But then I have upgraded the chromium to v 62 and more, then I got the compilation errors, related to the python:

`
| python ../../chrome/browser/resources/optimize_webui.py --host bookmarks --input ../../chrome/browser/resources/md_bookmarks --out_folder gen/chrome/browser/resources/md_bookmarks --depfile gen/chrome/browser/resources/md_bookmarks/build.d --html_in_files bookmarks.html --html_out_files vulcanized.html --js_out_files crisper.js
| ../../third_party/node/linux/node-linux-x64/bin/node ../../third_party/node/node_modules/polymer-bundler/lib/bin/polymer-bundler --exclude chrome://resources/html/polymer.html --exclude chrome://resources/polymer/v1_0/polymer/polymer.html --exclude chrome://resources/polymer/v1_0/polymer/polymer-micro.html --exclude chrome://resources/polymer/v1_0/polymer/polymer-mini.html --exclude chrome://resources/polymer/v1_0/web-animations-js/web-animations-next-lite.min.js --exclude chrome://resources/css/roboto.css --exclude chrome://resources/css/text_defaults.css --exclude chrome://resources/css/text_defaults_md.css --exclude chrome://resources/js/load_time_data.js --inline-css --inline-scripts --rewrite-urls-in-templates --strip-comments --redirect "chrome://resources/cr_components/|../../ui/webui/resources/cr_components" --redirect "chrome://resources/cr_elements/|../../ui/webui/resources/cr_elements" --redirect "chrome://resources/css/|../../ui/webui/resources/css" --redirect "chrome://resources/html/|../../ui/webui/resources/html" --redirect "chrome://resources/js/|../../ui/webui/resources/js" --redirect "chrome://resources/polymer/v1_0/|../../third_party/polymer/v1_0/components-chromium" --exclude strings.js --exclude chrome://bookmarks/strings.js --manifest-out /mnt/data/Yocto-miatech/yocto-miatech/build-apalis-imx6/tmp/work/cortexa9hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/63.0.3239.108-r0/chromium-63.0.3239.108/out/Release/gen/chrome/browser/resources/md_bookmarks/bookmarks_requestlist.txt --root /mnt/data/Yocto-miatech/yocto-miatech/build-apalis-imx6/tmp/work/cortexa9hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/63.0.3239.108-r0/chromium-63.0.3239.108/chrome/browser/resources/md_bookmarks --redirect "chrome://bookmarks/|/mnt/data/Yocto-miatech/yocto-miatech/build-apalis-imx6/tmp/work/cortexa9hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/63.0.3239.108-r0/chromium-63.0.3239.108/chrome/browser/resources/md_bookmarks" --out-dir gen/chrome/browser/resources/md_bookmarks/bundled --shell bookmarks.html --in-html bookmarks.html failed: /mnt/data/Yocto-miatech/yocto-miatech/build-apalis-imx6/tmp/work/cortexa9hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/63.0.3239.108-r0/chromium-63.0.3239.108/third_party/node/node_modules/polymer-bundler/lib/bin/polymer-bundler.js:178
|         const [prefix, path] = redirect.split('|');
|               ^
|
| SyntaxError: Unexpected token [
|     at exports.runInThisContext (vm.js:53:16)
|     at Module._compile (module.js:373:25)
|     at Object.Module._extensions..js (module.js:416:10)
|     at Module.load (module.js:343:32)
|     at Function.Module._load (module.js:300:12)
|     at Function.Module.runMain (module.js:441:10)
|     at startup (node.js:140:18)
|     at node.js:1043:3
|
| Traceback (most recent call last):
|   File "../../chrome/browser/resources/optimize_webui.py", line 252, in <module>
|     main(sys.argv[1:])
|   File "../../chrome/browser/resources/optimize_webui.py", line 235, in main
|     manifest_out_path = _optimize(args.input, args)
|   File "../../chrome/browser/resources/optimize_webui.py", line 159, in _optimize
|     ] + in_html_args)
|   File "../../third_party/node/node.py", line 28, in RunNode
|     raise
| TypeError: exceptions must be old-style classes or derived from BaseException, not NoneType

`

rakuco, 2019-06-24:
The Python part is a red herring, the error is in node.js. My guess is that your version is too old.

denis-shienkov, 2019-06-24:
> My guess is that your version is too old.
A version of what? Python?

rakuco, 2019-06-24:
> > My guess is that your version is too old.
> 
> A version of what? Python?

Of node.js.

rakuco, 2019-06-24:
Do note though that pyro as a branch is too old: it has GCC 6, which is too buggy and does not support enough C++14 to build Chromium successfully.

denis-shienkov, 2019-06-24:
> Of node.js.

Is it a host package, or it is a yocto package?

rakuco, 2019-06-24:
It's a Yocto package that runs on the host (i.e. we depend on `nodejs-native`). On newer branches it's part of meta-openembedded.

denis-shienkov, 2019-06-24:
Ok, many thanks.. 

But it is strange that chromium v63 does not compiled on 'pyro' branch, because from the 'meta-browser' commits log I see that it was checked on raspberry-pi && 'pyro' branch.

rakuco, 2019-06-24:
Ah, sorry, I didn't see you were building Chromium 63. I think that release still works with pyro and you can just find a newer nodejs recipe somewhere.

However, v63 is **very** old by Chromium's standards, and it is full of security issues. Please consider moving to the latest release we have here.

denis-shienkov, 2019-06-24:
Ok, many thanks, I will try to upgrade my yocto at least to 'rocko', and then, step by step more.. :)

denis-shienkov, 2019-06-25:
I just have updated the nodejs recipe to v8.4.0 (from 'rocko' branch) and then the chromium v63 compiled successfully. Many thanks.

rakuco, 2019-06-25:
Glad to hear that. Can we close this issue now?

denis-shienkov, 2019-06-25:
Yes, please.

closed 2019-06-25 by rakuco