OSSystems/meta-browser #280: chromium wayland commandbuffer error

raerne, 2019-07-01:
Hello

I finally got chromium to compile with the following setup
* chromium 73.0.3683.103 from commit cd88d7be61bc39726b8ecf7ddc19e6942bcf8cb4 (since the newer commits do not fix the wayland-scanner issues anymore)
* yocto rocko
* meta-ti layer for the am437x board

To be able to compile I had to set the following gn arguments
```
GN_ARGS += "\
        ${PACKAGECONFIG_CONFARGS} \
        use_ozone=true \
        ozone_auto_platforms=false \
        ozone_platform_headless=true \
        ozone_platform_wayland=true \
        ozone_platform_x11=false \
        system_wayland_scanner_path="${STAGING_BINDIR_NATIVE}/wayland-scanner" \
        use_xkbcommon=true \
        use_system_libwayland=true \
        use_system_minigbm=false \
        use_system_libdrm=true \
"
```
notice the `use_system_minigbm` is set to false, otherwise it didn't compile. Now running chromium as root gives me the following output
```
chromium --no-sandbox https://google.com
[5766:5796:0701/084453.601624:ERROR:bus.cc(396)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[5766:5766:0701/084453.878690:ERROR:desktop_window_tree_host_platform.cc(445)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::InitModalType(ui::ModalType)
[5766:5766:0701/084453.883558:ERROR:desktop_window_tree_host_platform.cc(455)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsAnimatingClosed() const
[5766:5766:0701/084453.890781:ERROR:desktop_window_tree_host_platform.cc(468)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SizeConstraintsChanged()
[5766:5766:0701/084454.097885:ERROR:desktop_window_tree_host_platform.cc(439)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SetWindowIcons(const gfx::ImageSkia&, const gfx::ImageSkia&)
[5766:5766:0701/084454.098377:ERROR:desktop_window_tree_host_platform.cc(371)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::SetWindowTitle(const string16&)
[5766:5766:0701/084454.562503:ERROR:desktop_window_tree_host_platform.cc(309)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Activate()
[5766:5766:0701/084454.625439:ERROR:desktop_window_tree_host_platform.cc(226)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsVisible() const
[5794:5794:0701/084455.125645:ERROR:in_process_command_buffer.cc(500)] ContextResult::kSurfaceFailure: Failed to create surface.
[5794:5806:0701/084455.771788:ERROR:wayland_canvas_surface.cc(89)] Not implemented reached in virtual std::unique_ptr<gfx::VSyncProvider> ui::WaylandCanvasSurface::CreateVSyncProvider()
[5808:5816:0701/084456.144714:ERROR:command_buffer_proxy_impl.cc(124)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
```

From issue #214 i see that i can fix this using the `--in-process-cpu` flag, however I am not sure how this affects (or disables) the hardware acceleration. Also the effect of the system/non-system gbm / minigbm / drm arguments during the build is a bit confusing to me.

Any help would be appreciated.
Ramon

rakuco, 2019-07-01:
Over to @msisov, who should be back in a couple of weeks. As for the wayland-scanner error, it's been fixed again in #278

msisov, 2019-07-15:
Hi,

if you use "use_system_minigbm=false", you have to specify exact driver to let the minigbm choose "right" implementation. https://cs.chromium.org/chromium/src/third_party/minigbm/BUILD.gn?q=use_system_minigbm&dr=C&l=16

msisov, 2019-07-29:
I suppose this can be closed

closed 2019-07-29 by msisov