OSSystems/meta-browser #281: V4L2 not work on NXP

guojing7, 2019-07-02:
Hi, I'm use NXP imx8qmmek to test hardware decoder through V4L2, but it not work.

You can see following chrome://media-internals messages, it first choose GpuVideoDecoder, but change to FFmpegVideoDecoder quickly, why? 

00:00:00.135	info	Selected FFmpegAudioDecoder for audio decoding, config: codec: vorbis, bytes_per_channel: 4, channel_layout: 3, channels: 2, samples_per_second: 48000, sample_format: 6, bytes_per_frame: 8, seek_preroll: 0us, codec_delay: 0, has extra data: true, encryption scheme: Unencrypted, discard decoder delay: true
00:00:00.249	video_dds	false
00:00:00.249	video_decoder	GpuVideoDecoder
00:00:00.249	is_platform_video_decoder	true
00:00:00.249	info	Selected GpuVideoDecoder for video decoding, config: codec: vp8, format: PIXEL_FORMAT_I420, profile: vp8, coded size: [1920,1080], visible rect: [0,0,1920,1080], natural size: [1920,1080], has extra data: false, encryption scheme: Unencrypted, rotation: 0°, color space: {primaries:BT709, transfer:BT709, matrix:BT709, range:LIMITED}
00:00:00.249	pipeline_state	kPlaying
00:00:00.251	warning	Failed to reconcile encoded audio times with decoded output.
00:00:00.258	audio_buffering_state	BUFFERING_HAVE_ENOUGH
00:00:00.302	video_dds	false
00:00:00.302	video_decoder	FFmpegVideoDecoder
00:00:00.302	is_platform_video_decoder	false
00:00:00.302	info	Selected FFmpegVideoDecoder for video decoding, config: codec: vp8, format: PIXEL_FORMAT_I420, profile: vp8, coded size: [1920,1080], visible rect: [0,0,1920,1080], natural size: [1920,1080], has extra data: false, encryption scheme: Unencrypted, rotation: 0°, color space: {primaries:BT709, transfer:BT709, matrix:BT709, range:LIMITED}

# Yocto
## Freescale BSP
repo init -u https://source.codeaurora.org/external/imx/imx-manifest -b imx-linux-sumo -m imx-4.14.98-2.0.0_ga.xml

## meta-browser
https://github.com/OSSystems/meta-browser.git
commit: f24753238414a8b04e5227772f22bb4dd9171b2f

# BUILD
Add following GN_ARGS to meta-browser/recipes-browser/chromium/chromium-ozone-wayland_75.0.3770.100.bb

GN_ARGS += "use_v4l2_codec=true use_v4lplugin=true use_linux_v4l2_only=true" 
GN_ARGS += 'proprietary_codecs=true ffmpeg_branding="Chrome"'

MACHINE=imx8qmmek DISTRO=fls-imx-wayland source fsl-setup-release.sh -b build-xwayland
bitbake fsl-image-qt5-validation-imx

# RUN
./chromium-wrapper --no-sandbox --in-process-gpu --ignore-gpu-blacklist --enable-logging=stderr --v=2

# chrome
## chrome://gpu
Graphics Feature Status
Canvas: Hardware accelerated
Flash: Hardware accelerated
Flash Stage3D: Hardware accelerated
Flash Stage3D Baseline profile: Hardware accelerated
Compositing: Hardware accelerated
Multiple Raster Threads: Enabled
Native GpuMemoryBuffers: Software only. Hardware acceleration disabled
Out-of-process Rasterization: Disabled
Hardware Protected Video Decode: Hardware accelerated
Rasterization: Software only. Hardware acceleration disabled
Skia Renderer: Disabled
Surface Control: Disabled
Surface Synchronization: Enabled
Video Decode: Hardware accelerated
Viz Service Display Compositor: Enabled
Viz Hit-test Surface Layer: Disabled
WebGL: Hardware accelerated
WebGL2: Hardware accelerated
Driver Bug Workarounds
clear_uniforms_before_first_program_use
exit_on_context_lost
scalarize_vec_and_mat_constructor_args
disabled_extension_GL_KHR_blend_equation_advanced
disabled_extension_GL_KHR_blend_equation_advanced_coherent
Problems Detected
Clear uniforms before first program use on all platforms: 124764, 349137
Applied Workarounds: clear_uniforms_before_first_program_use
Always rewrite vec/mat constructors to be consistent: 398694
Applied Workarounds: scalarize_vec_and_mat_constructor_args
Disable KHR_blend_equation_advanced until cc shaders are updated: 661715
[... 708 lines cut ...]
[12722:12813:0701/125411.298987:VERBOSE2:v4l2_video_decode_accelerator.cc(2232)] GetVisibleSize(): Fallback to VIDIOC_G_CROP
[12722:12813:0701/125411.299014:VERBOSE2:v4l2_video_decode_accelerator.cc(2246)] GetVisibleSize(): visible rectangle is 0,0 0x0
[12722:12813:0701/125411.299030:VERBOSE1:v4l2_video_decode_accelerator.cc(2253)] GetVisibleSize(): visible size is empty
[12722:12813:0701/125411.299049:VERBOSE2:v4l2_video_decode_accelerator.cc(2038)] StartResolutionChange(): Initiate resolution change
[12722:12744:0701/125411.299154:ERROR:gles2_cmd_decoder_autogen.h(143)] [.Media-0xffff700027b0]GL ERROR :GL_INVALID_ENUM : glBindTexture: target was GL_TEXTURE_EXTERNAL_OES
[12722:12744:0701/125411.299294:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff700027b0]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES
[12722:12744:0701/125411.299387:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff700027b0]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES
[12722:12744:0701/125411.299481:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff700027b0]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES
[12722:12744:0701/125411.299570:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff700027b0]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES
[12722:12744:0701/125411.300067:VERBOSE2:v4l2_video_decode_accelerator.cc(794)] Destroy():
[12722:12813:0701/125411.300397:VERBOSE2:v4l2_video_decode_accelerator.cc(1997)] StopOutputStream():
[12722:12813:0701/125411.300415:VERBOSE2:v4l2_video_decode_accelerator.cc(2599)] DestroyOutputBuffers():
[12722:12813:0701/125411.300426:VERBOSE2:v4l2_video_decode_accelerator.cc(2067)] FinishResolutionChange():
[12722:12813:0701/125411.300442:VERBOSE2:v4l2_video_decode_accelerator.cc(2232)] GetVisibleSize(): Fallback to VIDIOC_G_CROP
[12722:12813:0701/125411.300457:VERBOSE2:v4l2_video_decode_accelerator.cc(2246)] GetVisibleSize(): visible rectangle is 0,0 0x0
[12722:12813:0701/125411.300472:VERBOSE1:v4l2_video_decode_accelerator.cc(2253)] GetVisibleSize(): visible size is empty
[12722:12813:0701/125411.300483:VERBOSE2:v4l2_video_decode_accelerator.cc(2209)] CreateBuffersForFormat(): new resolution: 0x0, visible size: 0x0, decoder output planes count: 2, EGLImage size: 0x0, EGLImage plane count: 2
[12722:12813:0701/125411.300498:VERBOSE2:v4l2_video_decode_accelerator.cc(2540)] CreateOutputBuffers():
[12722:12813:0701/125411.300632:VERBOSE2:v4l2_video_decode_accelerator.cc(1916)] DestroyTask():
[12722:12813:0701/125411.300807:VERBOSE2:v4l2_video_decode_accelerator.cc(1997)] StopOutputStream():
[12722:12813:0701/125411.300826:VERBOSE2:v4l2_video_decode_accelerator.cc(2015)] StopInputStream():
[12722:12813:0701/125411.303026:VERBOSE2:v4l2_video_decode_accelerator.cc(2589)] DestroyInputBuffers():
[12722:12813:0701/125411.304806:VERBOSE2:v4l2_video_decode_accelerator.cc(2599)] DestroyOutputBuffers():
[12722:12744:0701/125411.305123:VERBOSE2:generic_v4l2_device.cc(505)] CloseDevice():
[12722:12744:0701/125411.307357:VERBOSE2:v4l2_video_decode_accelerator.cc(815)] Destroy(): Destroyed.

msisov, 2019-07-15:
[5262:5284:0701/050606.126705:VERBOSE1:generic_v4l2_device.cc(126)] SetDevicePollInterrupt(): write() failed: Bad file descriptor (9) .

Chromium uses different decoder device paths for ChromeOS, which are different from Linux. https://cs.chromium.org/chromium/src/media/gpu/v4l2/generic_v4l2_device.cc?q=generic_v4l2_device&dr=C&l=512

Make sure you provide correct links.

guojing7, 2019-07-16:
@msisov Thank you for your reply！

Yes, I already create  link /dev/video-dec to /dev/video12 (IMX VPU Decoder) before start chromium,  this error happen before open URL. So, i think it may be not the really problem, after open URL, never see this errors.

The GpuVideoDecoder destroyed seems by receive VideoDecodeAccelerator::INVALID_ARGUMENT:
https://cs.chromium.org/chromium/src/media/gpu/ipc/service/gpu_video_decode_accelerator.cc?q=gpu_video_decode_accelerator.cc&dr&l=455

Why? Is it related to the error log below?
[5262:5284:0701/050606.800456:ERROR:gles2_cmd_decoder_autogen.h(143)] [.Media-0xffff58002850]GL ERROR :GL_INVALID_ENUM : glBindTexture: target was GL_TEXTURE_EXTERNAL_OES
[5262:5284:0701/050606.800516:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff58002850]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES
[5262:5284:0701/050606.800557:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff58002850]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES
[5262:5284:0701/050606.800594:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff58002850]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES
[5262:5284:0701/050606.800628:ERROR:gles2_cmd_decoder_autogen.h(3038)] [.Media-0xffff58002850]GL ERROR :GL_INVALID_ENUM : glTexParameteri: target was GL_TEXTURE_EXTERNAL_OES