OSSystems/meta-browser #283: chromium-ozone-wayland no webpage rendered/displayed

lmajewski, 2019-07-05:
Dear Community,
I'd like to ask you for some help regarding running chromium-ozone-wayland on i.MX6Q.

* Setup:
  - HW: i.MX6Q with working etnaviv (kernel 4.18) and weston/wayland
  - YPRR: thud 
  - Host: debian-10 (buster) / gcc 8.3.0 (on HOST) - to avoid build issue with gcc < 7.2
  - meta-browser: SHA1: ab96b015dd086319def7a4e64124da67a1b0e0a0
    (I've also tried the version stable - 71 - SHA1: a7867648fa999be7db2bda2aa7adc18af5103a98, but results were pretty the same)
  - Used version: chromium-ozone-wayland_75.0.3770.100.bb

The only modification is with ```chromium-ozone-wayland_%.bbappend```

```
PACKAGECONFIG_pn-chromium-ozone-wayland = "\ 
                              use-egl \
                              impl-side-painting \ 
                              jumbo-build \
                              kiosk-mode \ 
                              proprietary-codecs \ 
                              use-linux-v4l2 \ 
                              " 
```

* SW configuration (after login in):
		-- mkdir -p /tmp/0-runtime-dir && chmod 0700 /tmp/0-runtime-dir
		-- export XDG_RUNTIME_DIR=/tmp/0-runtime-dir
		-- weston -B drm-backend.so -c /etc/weston.ini --tty=3 &
                   (glmark2-es2-wayland works without any issues or warnings)

* Starting Chromium ```chromium --no-sandbox wp.pl```

* Status
  - The chromium can be build without any issues
  - It can be started without problems.

* Problems:
  - For other wayland programs the ```/usr/libexec/weston-keyboard``` works. Unfortunately,
this is not the case with chromium (looks like it is not 'hooked' to it - is this protocol disabled?).
  - Unfortunately, it doesn't render/display any webpage - no matter if it is www.google.com or chrome://settings. Following [picture](https://commons.wikimedia.org/wiki/File:Chromium_47_Aw,_Snap!_screenshot_(en).png) is presented.
  - When I press "menu->settings" the chromium crashes with segfault. It seems like it is not related to webpage displaying problem (I will debug it latter).

* Logs:
```
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
[2611:2646:0701/131850.853645:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2611:2611:0701/131850.957235:ERROR:desktop_window_tree_host_platform.cc(446)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::InitModalType(ui::ModalType)
[2611:2611:0701/131850.957437:ERROR:desktop_window_tree_host_platform.cc(456)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsAnimatingClosed() const
[2611:2611:0701/131850.961684:ERROR:desktop_window_tree_host_platform.cc(469)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SizeConstraintsChanged()
[2611:2611:0701/131851.026024:ERROR:desktop_window_tree_host_platform.cc(440)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SetWindowIcons(const gfx::ImageSkia&, const gfx::ImageSkia&)
[2611:2611:0701/131851.026184:ERROR:desktop_window_tree_host_platform.cc(372)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::SetWindowTitle(const string16&)
[2611:2611:0701/131851.096780:ERROR:desktop_window_tree_host_platform.cc(310)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Activate()
[2611:2611:0701/131851.102510:ERROR:desktop_window_tree_host_platform.cc(227)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsVisible() const
Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x01c7463c
[2611:2621:0701/131851.199540:ERROR:service_manager_context.cc(242)] Attempting to run unsupported native service: /usr/lib/chromium/chrome_renderer.service
Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x01c7463c
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
[2710:2715:0701/131852.376157:ERROR:wayland_canvas_surface.cc(100)] Not implemented reached in virtual std::unique_ptr<gfx::VSyncProvider> ui::WaylandCanvasSurface::CreateVSyncProvider()
[2710:2715:0701/131852.438376:ERROR:wayland_canvas_surface.cc(106)] Not implemented reached in virtual void ui::WaylandCanvasSurface::OnSubmission(uint32_t, const gfx::SwapResult&)
[2710:2715:0701/131852.474970:ERROR:wayland_canvas_surface.cc(113)] Not implemented reached in virtual void ui::WaylandCanvasSurface::OnPresentation(uint32_t, const gfx::PresentationFeedback&)
[... 74 lines cut ...]
interface: 'weston_desktop_shell', version: 1, name: 19
interface: 'weston_screenshooter', version: 1, name: 20
```
* Output from ```WAYLAND_DEBUG=1 chromium --no-sandbox wp.pl > waylandlog.txt 2>&1;  grep -i error waylandlog.txt```
```
[3433:3468:0701/135158.705549:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[3433:3433:0701/135158.808095:ERROR:desktop_window_tree_host_platform.cc(446)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::InitModalType(ui::ModalType)
[3433:3433:0701/135158.808386:ERROR:desktop_window_tree_host_platform.cc(456)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsAnimatingClosed() const
[3433:3433:0701/135158.812595:ERROR:desktop_window_tree_host_platform.cc(469)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SizeConstraintsChanged()
[3433:3433:0701/135158.876101:ERROR:desktop_window_tree_host_platform.cc(440)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::SetWindowIcons(const gfx::ImageSkia&, const gfx::ImageSkia&)
[3433:3433:0701/135158.876400:ERROR:desktop_window_tree_host_platform.cc(372)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::SetWindowTitle(const string16&)
[3433:3433:0701/135158.975343:ERROR:desktop_window_tree_host_platform.cc(310)] Not implemented reached in virtual void views::DesktopWindowTreeHostPlatform::Activate()
[3433:3433:0701/135158.988643:ERROR:desktop_window_tree_host_platform.cc(227)] Not implemented reached in virtual bool views::DesktopWindowTreeHostPlatform::IsVisible() const
[3433:3443:0701/135159.080614:ERROR:service_manager_context.cc(242)] Attempting to run unsupported native service: /usr/lib/chromium/chrome_renderer.service
[3524:3529:0701/135200.190865:ERROR:wayland_canvas_surface.cc(100)] Not implemented reached in virtual std::unique_ptr<gfx::VSyncProvider> ui::WaylandCanvasSurface::CreateVSyncProvider()
[3524:3529:0701/135200.261017:ERROR:wayland_canvas_surface.cc(106)] Not implemented reached in virtual void ui::WaylandCanvasSurface::OnSubmission(uint32_t, const gfx::SwapResult&)
[3524:3529:0701/135200.291901:ERROR:wayland_canvas_surface.cc(113)] Not implemented reached in virtual void ui::WaylandCanvasSurface::OnPresentation(uint32_t, const gfx::PresentationFeedback&)
```

* Questions:
  - Can somebody confirm running this (or any other) version on i.MX6 device (or any other - like ```MACHINE="raspberrypi3"```) with properly displayed webpages?
  - Shouldn't it be possible to switch Chromium to pure SW rendering mode just to see if it works correctly (IIRC it is possible with X11)?
  - As chromium doesn't crash immediately with segfault, what might be the reason for not displaying/rendering webpages (there are a few problems in the log, so I'd like to prioritize errors to have first fixed displaying of the webpages)?

Thanks in advance for help (and patience :-) ).

msisov, 2019-07-15:
We haven't fixed gbm usage on imx yet. Please, use --in-process-gpu instead.

lmajewski, 2019-07-15:
Thanks Maksim for your reply.

The output of following command: ```chromium --in-process-gpu --no-sandbox wp.pl```
```
Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x01c35364
[823:833:0712/171431.228086:ERROR:service_manager_context.cc(242)] Attempting to run unsupported native service: /usr/lib/chromium/chrome_renderer.service
Segmentation fault
```

I'm wondering if the lack of chrome_renferer.service may be the culprit?

Anyway, shall I be aware of any other issues on i.MX (like if I'm missing any other switches to be enabled) before I start debugging this?

msisov, 2019-07-22:
I only know that dmabuf support is not available there. Otherwise, users did not have such a problem as you reported here.

msisov, 2021-03-18:
I guess this can be closed now.

closed 2021-03-18 by msisov