OSSystems/meta-browser #297: chromium: Update to 76.0.3809.100

rakuco, 2019-08-07:
This is a security fix release: 
https://chromereleases.googleblog.com/2019/08/stable-channel-update-for-desktop.html

Targets and platforms we need to check:

#### chromium-x11:
- [ ] thud, clang, MACHINE="intel-core2-32"
- [ ] thud, clang, MACHINE="intel-corei7-64"
- [ ] warrior, clang, MACHINE="raspberrypi3"

#### chromium-ozone-wayland:
- [ ] thud, clang, MACHINE="m3ulcb"
- [ ] thud, clang, MACHINE="qemuarm64"
- [ ] warrior, clang, MACHINE="raspberrypi3"

rakuco, 2019-08-07:
cc @msisov

rakuco, 2019-08-07:
It looks like the upstream tarball hasn't been generated yet though.

msisov, 2019-08-07:
@rakuco we also have the bot updated - https://github.com/Igalia/meta-browser/pull/32

rakuco, 2019-08-07:
We can, but looking at the "publish tarball" Chromium bot it will still take a while for the tarball to show up.

rakuco, 2019-08-10:
@msisov the tarball was finally published yesterday. Can you take the lead on this one? I think it's just a matter of updating the version number and the hashes, so most of the work is up to the build bots.

msisov, 2019-08-12:
Doing it.

PR #298 (2019-08-13): chromium: Update to 76.0.3809.100
Release announcement:
https://chromereleases.googleblog.com/2019/08/stable-channel-update-for-desktop.html

Test-built with
 ---------------
 * chromium-x11
   * thud, clang, MACHINE="intel-corei7-64"
   * thud, clang, MACHINE="intel-core2-32"
   * warrior, clang, MACHINE="intel-corei7-64"
 * chromium-ozone-wayland:
   * thud, clang, MACHINE="qemuarm64"
   * thud, clang, MACHINE="m3ulcb"
   * warrior, clang, MACHINE="raspberrypi3"

Fixes #297.
commit d0f2e64d (2019-08-13): chromium: Update to 76.0.3809.100

Release announcement:
https://chromereleases.googleblog.com/2019/08/stable-channel-update-for-desktop.html

Test-built with
 ---------------
 * chromium-x11
   * thud, clang, MACHINE="intel-corei7-64"
   * thud, clang, MACHINE="intel-core2-32"
   * warrior, clang, MACHINE="intel-corei7-64"
 * chromium-ozone-wayland:
   * thud, clang, MACHINE="qemuarm64"
   * thud, clang, MACHINE="m3ulcb"
   * warrior, clang, MACHINE="raspberrypi3"

Fixes #297.
closed 2019-08-13 by msisov