OSSystems/meta-browser #299: x11 in raspberry pi 4 chromium can not display on weida hi-tech cool touch screen in Yotco project

gitgonewithwind, 2019-08-20:
hi I open a n issue at here :
 https://github.com/agherzan/meta-raspberrypi/issues/467
maybe  the problem is from the chromium itself.
so I open at here. thank you to keep it. if I am wrong, please remove it.
I can not run chromium in raspberry pi4: ther version is chromium-x11_76.0.3809.100

 simply run : sudo -u myuser chromium in my old source code in pi3, works well.
but in current pi4 still it's Segmentation fault. I am going to add components one by one in local.conf to try.

raspberrypi4:/home/myuser# sudo -u myuser chromium  
Segmentation fault 

thanks

kraj, 2019-09-23:
Please try latest meta-clang branch ( master/warrior/thud ) whichever you are using and this issue should be fixed

gitgonewithwind, 2019-09-23:
@kraj  thanks. I am going to test.

gitgonewithwind, 2019-09-23:
@kraj 

I tried to build rasp4 32bit yocto, error. can not pass compilation. before compilation is ok. only can not ruu chromium on board. 

...
../../third_party/skia/src/gpu/ccpr/GrCoverageCountingPathRenderer.cpp -o yocto_native/obj/skia/skia/GrCoverageCountingPathRenderer.o
| ninja: build stopped: subcommand failed.
| WARNING: /opt/z/j-ws/touch-rpi4/build/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-x11/77.0.3865.90-r0/temp/run.do_compile.31363:1 exit 1 from 'ninja -v -j 10 chrome chrome_sandbox chromedriver'
| 
ERROR: Task (/opt/oe/configs/z/j-ws/touch-rpi4/layers/meta-browser/recipes-browser/chromium/chromium-x11_77.0.3865.90.bb:do_compile) failed with exit code '1'

gitgonewithwind, 2019-09-23:
@kraj 
raspberry p4 32bit yocto can not pass compilation:

ninja: build stopped: subcommand failed.
| WARNING: /opt/z/j-ws/touch-rpi4/build/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-x11/77.0.3865.90-r0/temp/run.do_compile.31363:1 exit 1 from 'ninja -v -j 10 chrome chrome_sandbox chromedriver'

gitgonewithwind, 2019-09-23:
@kraj 
master branch can not pass compilation for raspberry pi4 32bit : 

ninja: build stopped: subcommand failed.
| WARNING: /opt/z/j-ws/touch-rpi4/build/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-x11/77.0.3865.90-r0/temp/run.do_compile.31363:1 exit 1 from 'ninja -v -j 10 chrome chrome_sandbox chromedriver'

I am going to try warrior branch.

kraj, 2019-09-23:
@gitgonewithwind can you paste exact error where the compiler is failing?

gitgonewithwind, 2019-09-23:
@kraj 

/usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/unique_ptr.h:614:2: error: use of undeclared identifier 'swap'
|         swap(std::get<0>(_M_t), __p);
|         ^
| ../../third_party/skia/src/gpu/ccpr/GrCCFiller.cpp:215:22: note: in instantiation of function template specialization 'std::unique_ptr<GrTessellator::WindingVertex const[], std::default_delete<GrTessellator::WindingVertex const[]> >::reset<GrTessellator::WindingVertex *, void>' requested here
|     fFanTessellation.reset(vertices);
|                      ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/unique_ptr.h:626:7: note: must qualify identifier to find this declaration in dependent base class
|       swap(unique_ptr& __u) noexcept
|       ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/unique_ptr.h:614:26: error: too many arguments to function call, expected single argument '__u', have 2 arguments
|         swap(std::get<0>(_M_t), __p);
|         ~~~~                    ^~~
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/unique_ptr.h:625:7: note: 'swap' declared here
|       void
|       ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/unique_ptr.h:614:2: error: no matching function for call to 'swap'
|         swap(std::get<0>(_M_t), __p);
|         ^~~~
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/move.h:179:5: note: candidate template ignored: deduced conflicting types for parameter '_Tp' ('const GrTessellator::WindingVertex *' vs. 'GrTessellator::WindingVertex *')
|     swap(_Tp& __a, _Tp& __b)
|     ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/move.h:202:5: note: candidate template ignored: could not match '_Tp [_Nm]' against '__tuple_element_t<0UL, tuple<const GrTessellator::WindingVertex *, std::default_delete<GrTessellator::WindingVertex const[]> > >' (aka 'const GrTessellator::WindingVertex *')
|     swap(_Tp (&__a)[_Nm], _Tp (&__b)[_Nm])
|     ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/stl_pair.h:403:5: note: candidate template ignored: could not match 'pair<type-parameter-0-0, type-parameter-0-1>' against 'const GrTessellator::WindingVertex *'
|     swap(pair<_T1, _T2>& __x, pair<_T1, _T2>& __y)
|     ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/bits/basic_string.h:5287:5: note: candidate template ignored: could not match 'basic_string<type-parameter-0-0, type-parameter-0-1, type-parameter-0-2>' against 'const GrTessellator::WindingVertex *'
|     swap(basic_string<_CharT, _Traits, _Alloc>& __lhs,
|     ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/array:275:5: note: candidate template ignored: could not match 'array<type-parameter-0-0, _Nm>' against 'const GrTessellator::WindingVertex *'
|     swap(array<_Tp, _Nm>& __one, array<_Tp, _Nm>& __two)
|     ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/tuple:1546:5: note: candidate template ignored: could not match 'tuple<type-parameter-0-0...>' against 'const GrTessellator::WindingVertex *'
|     swap(tuple<_Elements...>& __x, tuple<_Elements...>& __y)
|     ^
| /usr/lib64/gcc/x86_64-suse-linux/6/../../../../include/c++/6/functional:2238:5: note: candidate template ignored: could not match 'function<type-parameter-0-0 (type-parameter-0-1...)>' against 'const GrTessellator::WindingVertex *'
|     swap(function<_Res(_Args...)>& __x, function<_Res(_Args...)>& __y)
|     ^
| 3 errors generated.

gitgonewithwind, 2019-09-23:
@kraj 

do I need add the  below  line to local.conf? I do  not know the meaning in README.md

----
LIBCPLUSPLUS_toolchain-clang_pn-<recipe> = ""

rakuco, 2019-09-23:
This looks like an issue building the parts of Chromium that run on the host. You seem to have GCC 6 installed, which is too old for Chromium; can you switch to a more recent version?

gitgonewithwind, 2019-09-23:
@rakuco 
 now it's compiling.,if can not pass in warrior branch. I am going to try higher gcc version.

gitgonewithwind, 2019-09-23:
@rakuco 
I did not try  raspberry pi4 64 bit with the updated meta-brower and meta-clang. now my company is using 64bit . it works well
I need make sure after I use higher gcc version . the raspberry pi4 64bit still works well. I need take time to try it. I mean if I use higher gcc version, if both 64bit and 32 bit can not work. I will waste my time to roll back.

gitgonewithwind, 2019-09-23:
warrior branch can not pass compilation. same error. for raspberry pi4 32bit

-----
I am going to try 64bit  
and then use high gcc version( I use current version for my projects for more than one years, I need make sure that my old project can work well using higher gcc  version)

gitgonewithwind, 2019-09-24:
same error for raspberry pi4 64 bit with the updated meta-brower and meta-clang.
I am going to try higher  gcc version.