OSSystems/meta-browser #302: Chromium-x11 failing at do_configure

klee4744, 2019-09-10:
Chromium-x11 build keep failing at do_configure 
Anyone have idea what's going on?

ERROR: chromium-x11-76.0.3809.132-r0 do_configure: Function failed: do_configure (log file is located at /home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/chromium-x11/76.0.3809.132-r0/temp/log.do_configure.56261)
ERROR: Logfile of failure stored in: /home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/chromium-x11/76.0.3809.132-r0/temp/log.do_configure.56261
Log data follows:
| DEBUG: Executing shell function do_configure
| /home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/chromium-x11/76.0.3809.132-r0/temp/run.do_configure.56261: 113: /home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/chromium-x11/76.0.3809.132-r0/temp/run.do_configure.56261: gn: not found
| WARNING: exit code 127 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/chromium-x11/76.0.3809.132-r0/temp/log.do_configure.56261)
ERROR: Task (/home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/layers/meta-openembedded/meta-browser/recipes-browser/chromium/chromium-x11_76.0.3809.132.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9626 tasks of which 9625 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

Summary: 1 task failed:
  /home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/layers/meta-openembedded/meta-browser/recipes-browser/chromium/chromium-x11_76.0.3809.132.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

rakuco, 2019-09-10:
> | [...]

This can mean two things: the `gn-native` recipe has not been built (in which case you shouldn't be able to get to chromium-x11's configuration stage in the first place), or the binary cannot be run on your host system for some reason.

Can you check if the binary is available, and try to run it on your host if it's there?

klee4744, 2019-09-10:
I can bitbake gn-native successfully.
How do you check if the binary is available or not?
Thank you.

rakuco, 2019-09-10:
You should be able to invoke the binary that's somewhere like `<work-dir>/tmp/work/x86_64-linux/gn-native/76.0.3809.132-r0/sysroot-destdir/[...]/usr/bin/gn` (the exact path can vary).

klee4744, 2019-09-10:
I have path up to /home/kwanl/openstlinux-4.19-thud-mp1-19-02-20/build-openstlinuxx11-stm32mp1/tmp-glibc/work/x86_64-linux/gn-native, but just a temp folder in this directory. Does that mean gn-native wasn't built successfully?
Thank you.

rakuco, 2019-09-11:
The easiest way to find that out is to try to build it again with `bitbake gn-native`. Your build directory should have more directories though. Here's what I have in mine:

```
$ ls ./work/x86_64-linux/gn-native/76.0.3809.132-r0
image  recipe-sysroot  recipe-sysroot-native  sysroot-destdir  temp  chromium-76.0.3809.132  0001-gen-Stop-passing-static-libstdc-to-the-compiler.patch
```

Are all recipes being built in the same machine?

AKADiNG, 2019-12-09:
> Chromium-x11 build keep failing at do_configure
> Anyone have idea what's going on?
>
> ```
> ERROR: chromium-x11-76.0.3809.132-r0 do_configure: Function failed: do_configure (log file is located at [...]

I had the same problem, and I find gn in chromium-x11's recipe-sysroot-native directory
core2-64-bst-linux/chromium-x11/78.0.3904.108-r0/recipe-sysroot-native/usr/bin/gn

Have you solve this problem, it is appreciate if you can share some experiences to me .

AndreyEroshenko, 2021-03-29:
Got the same problem when tried building chromium 83 (from zeus) using yocto-sumo on Ubuntu 20 host machine.
Problem with "gn" was caused by missing file "/lib/ld-linux-x86-64.so.2" in host filesystem. Fixed by adding symlink:
sudo ln -s /lib/x86_64-linux-gnu/ld-2.31.so /lib/ld-linux-x86-64.so.2