OSSystems/meta-browser #307: chromium: Update to 77.0.3865.90

rakuco, 2019-09-20:
https://chromereleases.googleblog.com/2019/09/stable-channel-update-for-desktop_18.html

Targets and platforms we need to check:

#### chromium-x11:
- [ ] thud, clang, MACHINE="intel-core2-32"
- [ ] thud, clang, MACHINE="intel-corei7-64"
- [ ] warrior, clang, MACHINE="raspberrypi3"
- [ ] warrior, clang, MACHINE="intel-corei7-64"

#### chromium-ozone-wayland:
- [ ] thud, clang, MACHINE="m3ulcb"
- [ ] thud, clang, MACHINE="qemuarm64"
- [ ] warrior, clang, MACHINE="raspberrypi3"

kraj, 2019-09-21:
I have fixed meta-clang /warrior for recently reported arm crash, please update and try

rakuco, 2019-09-21:
Does that change allow us to drop #305 ?

kraj, 2019-09-21:
that change is master meta clang specific but i think it belongs to meta-clang so i intend to drop it once i fix thud

kraj, 2019-09-21:
I have fixed thud meta-clang as well now chromium should build fine with thud as well.

rakuco, 2019-09-22:
> I have fixed thud meta-clang as well now chromium should build fine with thud as well.

Great, can you reference the changes in #299 and #304 and close them?

PR #310 (2019-09-23): chromium: Update to 77.0.3865.90
This is a security fix release:
https://chromereleases.googleblog.com/2019/09/stable-channel-update-for-desktop_18.html

Test-built with:

* chromium-x11
  * thud, MACHINE="intel-corei7-64"
  * thud, MACHINE="intel-core2-32"
  * warrior, MACHINE="intel-corei7-64"
  * warrior, MACHINE="intel-corei7-64"
* chromium-ozone-wayland:
  * thud, MACHINE="qemuarm64"
  * thud, MACHINE="m3ulcb"
  * warrior, MACHINE="raspberrypi3"

Fixes #307
closed 2019-09-23 by msisov