OSSystems/meta-browser #308: IMX6 Etnaviv Coda v4l2

mikeDonahoe, 2019-09-20:
I'm testing with chromium-ozone-wayland on the imx6 and have been unable to get Chromium to see the CODA decoder.  Am I missing something?

PACKAGECONFIG_append = " use-egl  proprietary-codecs use-linux-v4l2"

cat video7/name 
coda-decoder
lrwxrwxrwx    1 root     root            11 Sep 19 23:03 /dev/video-dec -> /dev/video7
lrwxrwxrwx    1 root     root            11 Sep 19 23:03 /dev/video-dec0 -> /dev/video7

LD_PRELOAD=/lib/libgcc_s.so.1 /usr/bin/chromium --ozone-platform=wayland --use-gl=egl --no-sandbox --ignore-gpu-blacklist

mikeDonahoe, 2019-09-20:
Looks like my issue is similar to #281 
[627:644:0920/154038.353183:VERBOSE2:v4l2_device.cc(27)] Create():
[627:644:0920/154038.353636:VERBOSE2:generic_v4l2_device.cc(152)] Initialize():
[627:644:0920/154038.358631:VERBOSE2:generic_v4l2_device.cc(494)] OpenDevicePath(): Using libv4l2 for /dev/video-dec
[627:644:0920/154038.358796:VERBOSE2:generic_v4l2_device.cc(502)] CloseDevice():
[627:644:0920/154038.359853:VERBOSE2:generic_v4l2_device.cc(502)] CloseDevice():
[627:644:0920/154038.360019:VERBOSE2:v4l2_device.cc(27)] Create():
[627:644:0920/154038.360110:VERBOSE2:generic_v4l2_device.cc(152)] Initialize():
[627:644:0920/154038.360218:VERBOSE2:v4l2_device.cc(27)] Create():
[627:644:0920/154038.360304:VERBOSE2:generic_v4l2_device.cc(152)] Initialize():
[627:644:0920/154038.360975:VERBOSE2:generic_v4l2_device.cc(502)] CloseDevice():
[627:644:0920/154038.361076:VERBOSE2:v4l2_video_encode_accelerator.cc(340)] Destroy():
[627:644:0920/154038.361168:VERBOSE2:v4l2_video_encode_accelerator.cc(543)] DestroyTask():
[627:644:0920/154038.361255:VERBOSE1:generic_v4l2_device.cc(128)] SetDevicePollInterrupt(): write() failed: Bad file descriptor (9)
[627:644:0920/154038.361599:VERBOSE2:v4l2_video_encode_accelerator.cc(146)] ~V4L2VideoEncodeAccelerator():
[627:644:0920/154038.361687:VERBOSE2:v4l2_video_encode_accelerator.cc(1369)] DestroyInputBuffers():
[627:644:0920/154038.361765:VERBOSE2:v4l2_video_encode_accelerator.cc(1389)] DestroyOutputBuffers():
[627:644:0920/154038.361856:VERBOSE2:generic_v4l2_device.cc(502)] CloseDevice():

mikeDonahoe, 2019-09-21:
Going through this still but not exactly like #281.  In EnumerateDevicesForType 
`buf_type = V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE;
`
EnumerateSupportedPixelformats returns nothing
Switching to V4L2_BUF_TYPE_VIDEO_OUTPUT gives me

> [1194:1211:0920/214640.483444:VERBOSE3:v4l2_device.cc(340)] EnumerateSupportedPixelformats(): Found H.264 (0x34363248)                                    
[1194:1211:0920/214640.483529:VERBOSE3:v4l2_device.cc(340)] EnumerateSupportedPixelformats(): Found MPEG-2 ES (0x3247504d)                                
[1194:1211:0920/214640.483606:VERBOSE3:v4l2_device.cc(340)] EnumerateSupportedPixelformats(): Found MPEG-4 part 2 ES (0x3447504d<

So it looks to be an issue with Single vs Multi-planar 

https://people.freedesktop.org/~narmstrong/meson_drm_doc/media/uapi/v4l/planar-apis.html

Ajay-Picus, 2020-06-26:
Hello Mike,

Are you able to get Chromium working with V4L2 decoder acceleration ? I am same issue which you have mentioned here.. Is this issue resolved?

msisov, 2021-03-18:
Probably, this can also be closed. Sorry for not being able to help you all with this.

closed 2021-03-18 by msisov