OSSystems/meta-browser #313: Chromium 77.0.3865.90 displaying no website on IMX6Q (ARM 32Bit)

dirkdatakamp, 2019-09-25:
Hi,

after switch to 77.0.3865.90  the browser didn't displaying any website. 

[image]

There are no error messages in the console and --disable-gpu option makes no difference.

Thanks
Dirk

rakuco, 2019-09-25:
Was this with or without #305? Which branch are you using, and have you checked you're using the latest meta-clang revision (which should fix the crash you were seeing before)? Was M76 working as expected?

dirkdatakamp, 2019-09-25:
I've tried with the latest meta-clang warrior. 

Currently I am compiling M76 with latest meta-clang warrior and without any further patches. But still takes about 2 hours.

dirkdatakamp, 2019-09-25:
Ok, 76.0.3809.132 works fine with the latest meta-clang warrior.
[image]

rakuco, 2019-09-25:
@msisov does this ring any bells since you've got more experience with ARM hardware?

msisov, 2019-09-25:
logs??? what does about://gpu say?

JeffyCN, 2019-09-29:
Hi @msisov ,

I can repro this issue on my rk3399 evb board(arm64). 

The crash log:
Received signal 11 SEGV_MAPERR 000000000000
#0 0x00555d237154 base::debug::CollectStackTrace()
#1 0x00555d19e7cc base::debug::StackTrace::StackTrace()
#2 0x00555d23711c base::debug::(anonymous namespace)::StackDumpSignalHandler()
#3 0x007f91d9e6c0 ([vdso]+0x6bf)
#4 0x007f8f795578 std::__cxx11::basic_string<>::_M_assign()
#5 0x00555f9a21d0 blink::FontCacheKey::operator=()
#6 0x00556122807c _ZN3WTF9HashTableIN5blink12FontCacheKeyENS_12KeyValuePairIS2_13s
coped_refptrINS1_17SegmentedFontDataEEEENS_24KeyValuePairKeyExtractorENS1_16FontCa
cheKeyHashENS_18HashMapValueTraitsINS1_18FontCacheKeyTraitsENS_10HashTraitsIS6_EEE
ESB_NS_18PartitionAllocatorEE6insertINS_17HashMapTranslatorISE_S9_SF_EERS2_DnEENS_
18HashTableAddResultISG_S7_EEOT0_OT1_
#7 0x005561226668 blink::CSSSegmentedFontFace::GetFontData()
#8 0x00555f9a97a4 blink::FontFallbackList::GetFontData()
#9 0x00555f9a9628 blink::FontFallbackList::FontDataAt()
#10 0x00555f9a9394 blink::FontFallbackList::DeterminePrimarySimpleFontData()
#11 0x00556044e814 blink::ComputedStyle::ComputedLineHeight()
#12 0x00555ff9d5e0 blink::TextControlInnerEditorElement::CreateInnerEditorStyle()
#13 0x00555ff9d280 blink::TextControlInnerEditorElement::CustomStyleForLayoutObjec
t()
#14 0x00555fc924f4 blink::Element::StyleForLayoutObject()
#15 0x00555fc92c10 blink::Element::RecalcOwnStyle()
#16 0x00555fc92720 blink::Element::RecalcStyle()
#17 0x00555fc41f08 blink::ContainerNode::RecalcDescendantStyles()
#18 0x00555fcedca0 blink::ShadowRoot::RecalcStyle()
#19 0x00555fc92824 blink::Element::RecalcStyle()
#20 0x00555fc41f08 blink::ContainerNode::RecalcDescendantStyles()
#21 0x00555fc92864 blink::Element::RecalcStyle()
#22 0x00555fc41f08 blink::ContainerNode::RecalcDescendantStyles()
#23 0x00555fc92864 blink::Element::RecalcStyle()
#24 0x00555fc41f08 blink::ContainerNode::RecalcDescendantStyles()
#25 0x00555fc92864 blink::Element::RecalcStyle()
#26 0x00555fc41f08 blink::ContainerNode::RecalcDescendantStyles()
#27 0x00555fc92864 blink::Element::RecalcStyle()
#28 0x00555fc41f08 blink::ContainerNode::RecalcDescendantStyles()
#29 0x00555fc92864 blink::Element::RecalcStyle()
#30 0x00555fc19f64 blink::StyleEngine::RecalcStyle()
#31 0x00555fc51d54 blink::Document::UpdateStyle()
#32 0x00555fc4d7dc blink::Document::UpdateStyleAndLayoutTree()
#33 0x00555fe701c8 blink::LocalFrameView::UpdateStyleAndLayoutIfNeededRecursive()
#34 0x00555fe6f98c blink::LocalFrameView::RunStyleAndLayoutLifecyclePhases()
#35 0x00555fe6f44c blink::LocalFrameView::UpdateLifecyclePhasesInternal()
#36 0x00555fe6e9c4 blink::LocalFrameView::UpdateLifecyclePhases()
#37 0x00556033d7a4 blink::PageAnimator::UpdateAllLifecyclePhases()
#38 0x00555fdd24d4 blink::WebViewImpl::UpdateLifecycle()
#39 0x0055607495d4 content::RenderWidget::UpdateVisualState()
#40 0x00555e2c78a0 cc::ProxyMain::BeginMainFrame()
#41 0x00555e31d504 base::internal::Invoker<>::RunOnce()
#42 0x00555d1ec730 base::TaskAnnotator::RunTask()
#43 0x00555d1fc268 base::sequence_manager::internal::ThreadControllerWithMessagePu
mpImpl::DoWorkImpl()
#44 0x00555d1fbfc0 base::sequence_manager::internal::ThreadControllerWithMessagePu
mpImpl::DoSomeWork()
#45 0x00555d1b5ba4 base::MessagePumpDefault::Run()
#46 0x00555d1fcb80 base::sequence_manager::internal::ThreadControllerWithMessagePu
mpImpl::Run()
#47 0x00555d1d41e4 base::RunLoop::RunWithTimeout()
#48 0x005560758b6c content::RendererMain()
#49 0x00555cd48e14 content::RunZygote()
#50 0x00555cd49b04 content::ContentMainRunnerImpl::Run()
#51 0x00555cd706dc service_manager::Main()
#52 0x00555cd48480 content::ContentMain()
#53 0x00555a969f54 ChromeMain
#54 0x007f8f52ece4 __libc_start_main
#55 0x00555a969de4 <unknown>
[end of stack trace]
Calling _exit(1). Core file will not be generated.

JeffyCN, 2019-09-30:
FYI, this patch fix the issue:
https://chromium.googlesource.com/chromium/src/+/74138b9febd37eac0fc26b8efb110014a83a52c6%5E%21/

msisov, 2019-09-30:
Heh. Just started to look into it and here goes your comment. nice! thanks for working on this as well

We will backport it

rakuco, 2019-09-30:
@dirkdatakamp @JeffyCN can you verify if #315 fixes the crash you were experiencing?

dirkdatakamp, 2019-10-01:
@rakuco works fine. Thanks guy's :)

rakuco, 2019-10-01:
Closing, fixed by #315.

closed 2019-10-01 by rakuco