OSSystems/meta-browser #314: Chromium 77.0.3865.90 component build broken

JeffyCN, 2019-09-30:
Hi guys,

Tested on rk3399 evb board(arm64), the component build is broken now.

1/ libmedia_gpu.so failed to link, which is because v4l2_jpeg_encode_accelerator.cc depends on chromeos_camera:
https://chromium.googlesource.com/chromium/src/+blame/HEAD/media/gpu/v4l2/BUILD.gn#102

This patch works:
+++ b/media/gpu/v4l2/BUILD.gn
@@ -32,8 +32,6 @@ source_set("v4l2") {
     "v4l2_device.h",
     "v4l2_image_processor.cc",
     "v4l2_image_processor.h",
-    "v4l2_jpeg_encode_accelerator.cc",
-    "v4l2_jpeg_encode_accelerator.h",
     "v4l2_video_decode_accelerator.cc",
     "v4l2_video_decode_accelerator.h",
     "v4l2_video_encode_accelerator.cc",
@@ -94,6 +92,8 @@ source_set("v4l2") {

   if (is_chromeos) {
     sources += [
+      "v4l2_jpeg_encode_accelerator.cc", <--- this depends on chromeos_camera
+      "v4l2_jpeg_encode_accelerator.h",

2/ Failed to generate v8 bin:
out/Release/libfreetype_harfbuzz.so: undefined symbol: uscript_getShortName_64

Which is because the generated libfreetype_harfbuzz.so depends on generated libicuuc.so, and we override LD_LIBRARY_PATH to prefer searching STAGING_DIR_HOST instead of output dir in:
https://github.com/OSSystems/meta-browser/blob/master/recipes-browser/chromium/chromium-gn.inc#L311

Maybe we should add $(B) (output dir) in front of the LD_LIBRARY_PATH.

rakuco, 2020-03-23:
@JeffyCN is everything working fine now?

JeffyCN, 2020-03-23:
> @JeffyCN is everything working fine now?

No...The first error is fixed in newer versions, but the second one still exists(conflict when linking the system lib instead of chromium generated ones).

And i hit more errors when trying component-build in r80(missing headers)...

rakuco, 2020-03-23:
OK, I'll keep this open then so we can figure out whether it makes sense to fix all those errors or just drop support for component builds altogether.

cc @msisov

JeffyCN, 2020-03-24:
FYI, the newest error should be fixed in:
https://pdfium.googlesource.com/pdfium.git/+/93101066fa60961b58324276165a2de6126e0f5d%5E%21/BUILD.gn