OSSystems/meta-browser #316: chromium-ozone-wayland wayland-ime keycode error

JeffyCN, 2019-10-08:
Hi guys,

I've been reported when using wayland-ime sometimes some special keys doesn't work as expected, for example enter key becomes '=', tab key becomes '8'.

This is because the keycode conversion in chromium is wrong(using virtual keycode as evdev keycode), this patch works:
https://github.com/JeffyCN/meta-rockchip/blob/master/dynamic-layers/recipes-browser/chromium/chromium-ozone-wayland_77.0.3865/0005-ui-ozone-wayland-wayland-ime-Fix-keycode-conversion.patch 

Since it's a chromium related issue, i'll report to them too.

JeffyCN, 2019-10-08:
chromium issue opened at:
https://bugs.chromium.org/p/chromium/issues/detail?id=1012160

msisov, 2019-10-09:
tracking it in crbug instead.

closed 2019-10-09 by msisov