OSSystems/meta-browser #320: chromium: Update to 78.0.3904.70

rakuco, 2019-10-23:
https://chromereleases.googleblog.com/2019/10/stable-channel-update-for-desktop_22.html

This is a new milestone, so there's probably some additional work compared to a regular, minor update.

Targets and platforms we need to check:

#### chromium-x11:
- [ ] thud, clang, MACHINE="intel-core2-32"
- [ ] thud, clang, MACHINE="intel-corei7-64"
- [ ] warrior, clang, MACHINE="raspberrypi3"
- [ ] warrior, clang, MACHINE="intel-corei7-64"

#### chromium-ozone-wayland:
- [ ] thud, clang, MACHINE="m3ulcb"
- [ ] thud, clang, MACHINE="qemuarm64"
- [ ] warrior, clang, MACHINE="raspberrypi3"

msisov, 2019-10-23:
Hey, I’m on holidays and will be back on 4th of November. If it can wait until then, I’ll complete the task once back

rakuco, 2019-10-29:
I've started the work in my m78-70 branch.

msisov, 2019-11-04:
Hey. I'm back. I'll test wayland build targets

akarshanbiswas, 2019-11-05:
Update to 78.0.3904.87 directly. There is a zero day security vulnerability. 
https://chromereleases.googleblog.com/2019/10/stable-channel-update-for-desktop_31.html?m=1

msisov, 2019-11-05:
triggered the bot to test the m78-87

closed 2019-11-06 by msisov