OSSystems/meta-browser #323: Red

mawildoer, 2019-10-27:


mawildoer, 2019-10-28:
Hi all, 

We're running into an issue with the browser starting where it fills the entire frame buffer with red. 

Issue with commits: 
 - 2132e3e45b972fd3fa8b28016603e8e14dd5565a (chromium 69.0.3497.100?)
 - 8ab83eda5227edad45f9a907042e7f5b94d6ed2d (chromium 77.0.3865.120)

Wasn't an issue with d807b901b0b68b478093964bd60a77b9376b3f12

Has anyone else seen this issue? 

It's the entire screen of chrome, including the tab bar etc... 
The other program we have to test against is psplash, which didn't have the same issue on startup. 

I have attached a snippet of video for reference: https://photos.app.goo.gl/eEh4x7TSaECvWjM87

Thanks for any assistance!

Matt

rakuco, 2019-10-29:
Hi. Are you running chromium-x11 or chromium-ozone-wayland? What hardware/GPU do you have?

mawildoer, 2019-10-30:
Sorry, for the lack of detail originally.

We're running chromium-x11 on an NXP/Freescale i.MX6UltraLite which doesn't have a GPU. 

It almost seems to be chrome://flags/#tint-gl-composited-content, but only on startup and we aren't using this tag (explicitly/intentionally at least). It also reads as disabled in the chrome flags menus. 

Have you got any ideas? :)

fabianbuettner, 2020-03-31:
I am also using chromium-x11 on a i.MX6ULL-EVK.

I noticed that the red background is only visible if the x server is using a 16-Bit framebuffer.
After setting DefaultDepth 24 in the xorg.conf I get a grey frame while the application is starting.