OSSystems/meta-browser #332: Build failure on Poky zeus on aarch64

superna9999, 2019-11-22:
Current setup :
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "seirobotics-sei610"
DISTRO               = "poky"
DISTRO_VERSION       = "3.0"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:0275e68b0576cdefc19718e4b1977044f9f43f48"
meta-oe              = "HEAD:aad5b3d070cd8c58828b0975cf861d8ebc90f460"
meta-clang           = "master:5eec6f9208a410767a918fd9beadb39620430811"
meta-browser         = "HEAD:22139d52f902f13d8433e3458858ea230eaaadfe"
meta-meson           = "topic/g12a/integ:e251fdee3ff995acab77a0f3dc550775877dcdee"

Build fails with:
MACHINE=seirobotics-sei610 bitbake chromium-ozone-wayland

```
| FAILED: yocto_native/obj/base/base/cpu.o
| clang++ -MMD -MF yocto_native/obj/base/base/cpu.o.d -DUSE_SYMBOLIZE -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS
=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVAL
GRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DBASE_IMPLEMENTATION -I../.. -Iyocto_native/gen -I../../third_party/boringssl/src/include -fno-strict-aliasing --param=ssp-
buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-di
r=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -m64 -march=x86-64 -Wall -Wextra -Wimplicit-fallthrough -Wthread-safety -
Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored
-pragma-optimize -Wno-implicit-int-float-conversion -Wno-xor-used-as-pow -Wno-c99-designator -Wno-reorder-init-list -Wno-final-dtor-non-final-class -fno-omit-fram
e-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wglobal-constructors -Wexit-time-destructors -Wshadow -Wexi
t-time-destructors -O2 -fno-ident -fdata-sections -ffunction-sections -std=c++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/home/narmstrong/pr
ojects/amlogic/yocto/build-thud-g12a/tmp/work/aarch64-poky-linux/chromium-ozone-wayland/78.0.3904.97-r0/recipe-sysroot-native/usr/include -isystem/home/narmstrong
/projects/amlogic/yocto/build-thud-g12a/tmp/work/aarch64-poky-linux/chromium-ozone-wayland/78.0.3904.97-r0/recipe-sysroot-native/usr/include -O2 -pipe -c ../../ba
se/cpu.cc -o yocto_native/obj/base/base/cpu.o
| warning: unknown warning option '-Wno-implicit-int-float-conversion'; did you mean '-Wno-implicit-float-conversion'? [-Wunknown-warning-option]
| warning: unknown warning option '-Wno-xor-used-as-pow'; did you mean '-Wno-unused-macros'? [-Wunknown-warning-option]
| warning: unknown warning option '-Wno-c99-designator'; did you mean '-Wno-gnu-designator'? [-Wunknown-warning-option]
| warning: unknown warning option '-Wno-reorder-init-list'; did you mean '-Wno-empty-init-stmt'? [-Wunknown-warning-option]
| warning: unknown warning option '-Wno-final-dtor-non-final-class'; did you mean '-Wno-abstract-final-class'? [-Wunknown-warning-option]
| ../../base/cpu.cc:62:10: error: chosen constructor is explicit in copy-initialization
|   return {family, model, ext_family, ext_model};
|          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/tuple:479:19: note: explicit constructor declared here
|         constexpr tuple(_UElements&&... __elements)
|                   ^
| 5 warnings and 1 error generated.
```

Full chromium build log: https://termbin.com/6ifd

rakuco, 2019-11-22:
> | FAILED: yocto_native/obj/base/base/cpu.o
> | /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/tuple:479:19: note: explicit constructor declared here

Your host toolchain that's used to build some parts that run during the build is too old. You'd probably need at least libstdc++ 7, not 5.

JeffyCN, 2019-11-26:
Update your PC's gcc from 5.4.0 to 6.x should help, the support of gcc 5 has been removed at:
https://github.com/OSSystems/meta-browser/commit/1bfc47f211b5eecc40fa14457b1867528648d756

See the comment inside, "  GCC 5 does not accept the list-initialization syntax for std::tuple."

rakuco, 2019-11-26:
Note that by now I think even GCC 6 is too old (see 50b56b085c3eacf4e3bbabc246d45dc75c708d42, for example).
Depending on demand, I guess we could try to build the native bits with clang and libc++ too.

rakuco, 2021-12-16:
@superna9999 is this still an issue? We haven't supported zeus in a long time and our toolchain requirements have changed, so I'm wondering if everything is working fine for you now.

superna9999, 2021-12-16:
I honestly haven't tried since, so it may be fixed.

closed 2021-12-16 by superna9999