OSSystems/meta-browser #333: When trying to click a <select> element in the page, xdg_pop() fails

jkarnataki, 2019-11-26:
chromium_ozone_wayland version 75.0.3770.142

In any HTML page which has a <select> element, touching the select element should bring a popup, but there is a DCHECK failure at:

ui/ozone/platform/wayland/host/wayland_window.cc
DCHECK(parent_window_ && !xdg_popup_);

rakuco, 2019-11-26:
Chromium 75 is pretty old by Chromium standards, and especially so by chromium-ozone-wayland standards. Can you reproduce this with the latest release in this tree?

msisov, 2019-12-02:
please reopen if the issue still exists in the latest release.

closed 2019-12-02 by msisov

jkarnataki, 2019-12-05:
Issue is reproducible in 78.0.3904.108.

msisov, 2019-12-05:
actually, we have this bug reported in the crbug. We'll work on it as soon as possible.