OSSystems/meta-browser #335: When trying to click a <select> element in the page, xdg_pop() fails

jkarnataki, 2019-12-05:
(This is actually continuation of #333. Sorry, cannot try the latest version on the target, but have taken the latest one which works on the target)
chromium_ozone_wayland version 78.0.3904.108

In any HTML page which has a select element, touching the select element should bring a popup, but there is a DCHECK failure at:

ui/ozone/platform/wayland/host/wayland_window.cc
WaylandWindow::CreateXdgPopup()
Line 238: DCHECK(parent_window_ && !xdg_popup_);

fichter, 2019-12-05:
Possibly related to [Issue 973015: Touch event on select tag crashes browser](https://bugs.chromium.org/p/chromium/issues/detail?id=973015)?

msisov, 2021-03-18:
Closing as https://bugs.chromium.org/p/chromium/issues/detail?id=973015 already track it.

closed 2021-03-18 by msisov