OSSystems/meta-browser #344: chromium-ozone-wayland gpu video decoding is broken

JeffyCN, 2019-12-18:
Hi guys,

The gpu video decoding is broken since:
https://chromium.googlesource.com/chromium/src/+/b9accafbaface80122defbc347257f528b4e6984%5E%21/media/renderers/default_decoder_factory.cc
-      video_decoders->push_back(std::make_unique<GpuVideoDecoder>(
-          gpu_factories, request_overlay_info_cb, target_color_space,
-          media_log));

https://chromium.googlesource.com/chromium/src/+/e2277737669db4d2a442a0bd8c07dd09a8c1fbb2
[media] Remove GpuVideoDecoder.

Should we add GpuVideoDecoder back or turn to use mojo decoder?

msisov, 2019-12-18:
I don't believe this is the right place to ask. Please report to upstream via crbug.com

JeffyCN, 2019-12-18:
> I don't believe this is the right place to ask. Please report to upstream via crbug.com

Hmm, actually the upstream chromium doesn's support vda on linux, the related patches are maintained here...

The upstream seems turn to use mojo decoder now, and for linux it's only enabled with vaapi, i'll try to test it

JeffyCN, 2019-12-19:
Hi guys,

Enabling mojo media make v4l2 codec works again(tested on rk3399 evb platform).

--- a/media/media_options.gni
+++ b/media/media_options.gni
@@ -150,7 +150,7 @@ declare_args() {
   enable_mojo_media =
       !is_fuchsia &&
       (is_android || is_chromeos || is_mac || is_win || enable_library_cdms ||
-       (is_desktop_linux && use_vaapi) || is_chromecast)
+       (is_desktop_linux && (use_v4l2_codec || use_vaapi)) || is_chromecast)

   # Enable the TestMojoMediaClient to be used in mojo MediaService. This is for
   # testing only and will override the default platform MojoMediaClient, if any.
@@ -209,7 +209,7 @@ if (enable_mojo_media) {
     ]
     _default_mojo_media_host = "gpu"
   } else if (is_chromeos || is_mac || is_win ||
-             (is_desktop_linux && use_vaapi)) {
+             (is_desktop_linux && (use_v4l2_codec || use_vaapi))) {
     _default_mojo_media_services = [ "video_decoder" ]
     _default_mojo_media_host = "gpu"
   }

Should we squash that into this patch?(maybe also test it on some other platforms)
https://github.com/OSSystems/meta-browser/blob/master/recipes-browser/chromium/chromium-ozone-wayland/V4L2/0001-Add-support-for-V4L2VDA-on-Linux.patch

@rakuco  @msisov

msisov, 2019-12-19:
Actually, I planned to remove V4L2 patches during next update of the recipe (I have already created a PR) as long as we have not been testing it for many months, but still have been rebasing it and shipping it within the recipe. I believe it makes 0 sense to ship something, which is never tested. However, if you're interested in V4L2 support, you may take the patches and keep them in your repository... Or become the maintainer of V4L2 here in this repo.

JeffyCN, 2019-12-19:
Okay, that make sense, i'll keep the patches in my repo, thanks :)

closed 2019-12-19 by JeffyCN