OSSystems/meta-browser #346: chromium-ozone-wayland: 79.0.3945.79: Failed to apply 0001-Fix-building-with-pulseaudio-13.patch

JeffyCN, 2019-12-23:
Hi guys,

When building the newest meta-browser's chromium-ozone-wayland:
```
ERROR: chromium-ozone-wayland-79.0.3945.79-r0 do_patch: Command Error: 'quilt --quiltrc /home/jeffy/android/sata/yocto/test/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/chromium-ozone-wayland/79.0.3945.79-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch 0001-Fix-building-with-pulseaudio-13.patch
patching file media/audio/pulse/pulse.sigs
Hunk #1 FAILED at 24.
Hunk #2 FAILED at 38.
2 out of 2 hunks FAILED -- rejects in file media/audio/pulse/pulse.sigs
patching file media/audio/pulse/pulse_stub_header.fragment
Hunk #1 FAILED at 5.
1 out of 1 hunk FAILED -- rejects in file media/audio/pulse/pulse_stub_header.fragment
Patch 0001-Fix-building-with-pulseaudio-13.patch can be reverse-applied
ERROR: Logfile of failure stored in: /home/jeffy/android/sata/yocto/test/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/chromium-ozone-wayland/79.0.3945.79-r0/temp/log.do_patch.28759
ERROR: Task (/home/jeffy/android/sata/yocto/test/build/../meta-browser/recipes-browser/chromium/chromium-ozone-wayland_79.0.3945.79.bb:do_patch) failed with exit code '1'
```

Which seems because the recipes-browser/chromium/files/0001-Fix-building-with-pulseaudio-13.patch has already landed in upstream's 79.0.3945.79:
https://chromium.googlesource.com/chromium/src/+blame/refs/tags/79.0.3945.79/media/audio/pulse/pulse.sigs#31

kraj, 2019-12-23:
I guess both patches to upgrade chromium and pulse audio landed same time

JeffyCN, 2019-12-23:
> I guess both patches to upgrade chromium and pulse audio landed same time

Likely, and the same as the delete_not_yet_released_clang_warnings.patch.

PR #347 (2020-01-04): chromium: Forward port warning patch and delete pa13 patch
pa13 patch is already present in latest chromium
Forward port delete_not_yet_released_clang_warnings.patch to apply on
latest chromium

Fixes #346
closed 2020-01-04 by rakuco