OSSystems/meta-browser #349: chromium: update to 79.0.3945.117

msisov, 2020-01-13:
https://chromereleases.googleblog.com/2020/01/stable-channel-update-for-desktop.html

Targets and platforms we need to check:

#### chromium-x11:
- [ ] thud, clang, MACHINE="intel-core2-32"
- [ ] thud, clang, MACHINE="intel-corei7-64"
- [ ] warrior, clang, MACHINE="raspberrypi3"
- [ ] warrior, clang, MACHINE="intel-corei7-64"

#### chromium-ozone-wayland:
- [ ] thud, clang, MACHINE="m3ulcb"
- [ ] thud, clang, MACHINE="qemuarm64"
- [ ] warrior, clang, MACHINE="raspberrypi3"

msisov, 2020-01-13:
https://gitlab.com/browsers/meta-browser/pipelines/108844750

kraj, 2020-01-13:
I wonder if we should drop thud and start testing zeus instead ?

msisov, 2020-01-16:
Mmm, probably yes. Do you know for how long thud is going to be a stable branch?

PR #350 (2020-01-16): chromium: update to 79.0.3945.117
See https://chromereleases.googleblog.com/2020/01/stable-channel-update-for-desktop.html

Test-built with:
* chromium-x11
  * thud, MACHINE="intel-corei7-64"
  * thud, MACHINE="intel-core2-32"
  * warrior, MACHINE="intel-corei7-64"
  * warrior, MACHINE="intel-corei7-64"
* chromium-ozone-wayland:
  * thud, MACHINE="qemuarm64"
  * thud, MACHINE="m3ulcb"
  * warrior, MACHINE="raspberrypi3"

Fixes: #349
closed 2020-01-16 by msisov