OSSystems/meta-browser #352: chromium-x11: Found local symbols on part of the global table

aehs29, 2020-01-21:
Hello,
While I'm not entirely sure this is a bug (since I figured you would have run into this already), I am wondering if anyone has run into this problem:

| clang++ -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse -ld=lld -Wl,--icf=all -Wl,--color-diagnostics -m64 -Wl,-O2 -Wl,--gc-sections -rdynamic -pie -Wl,--disable-new-dtags -Lbuilds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib -Lbuilds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib  -Lbuilds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/lib -Wl,--enable-new-dtags -Wl,-rpath-link,builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib  -Wl,-rpath-link,builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/lib -Wl,-rpath,builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/lib -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=builds/pixelbook/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -o "yocto_native/transport_security_state_generator" -Wl,--start-group @"yocto_native/transport_security_state_generator.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4

| ld.lld: error: found local symbol 'NSS_3.9.3' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.8' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.4' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.6' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.2' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.9' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.15' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.13' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.7' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.3' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.2.1' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.18' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.10' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| ld.lld: error: found local symbol 'NSS_3.12.10' in global part of symbol table in file builds/pixelbook/tmp/work/skylake-64-poky-linux/chromium-x11/79.0.3945.117-r0/recipe-sysroot-native/usr/lib/libsmime3.so
| clang-9: error: linker command failed with exit code 1 (use -v to see invocation)

My initial guess was that my toolchain might be too old, but I have already upgraded it with no luck.

Thanks in advance.
Alejandro

rakuco, 2020-01-21:
Hi, what distro are you building this on? The errors come from a binary that is built and linked on the host, so I'm still not sure if that's because your host toolchain is too old or too new actually.

kraj, 2020-01-21:
perhaps disable using lld linker on host. If you have `ld-is-lld` `DISTRO_FEATURES` then reset it.

or meta-clang has a bbappend which enforces lld perhaps comment that out

`dynamic-layers/browser-layer/recipes-browser/chromium-x11_%.bbappend`
```
GN_ARGS += "use_lld=true use_gold=false"
```

aehs29, 2020-01-23:
Hey Khem,
I had already tried changing the DISTRO_FEATURES, I did not see that meta-clang was enforcing that, I'm gonna give that a try now.

Alejandro

aehs29, 2020-01-24:
Hey guys,
I was able to build now, commenting that line on meta-clang fixed my linking issue.

Still, I'd like to understand what part(s) of the Host toolchain are required to be in a certain version?, it is just GCC and libstdc++?, binutils as well? anything besides that?

Thanks

Oquirella, 2023-02-08:
I have the same problem, it is still not fixed.