OSSystems/meta-browser #353: gn-native failing to build on warrior

jpsalm, 2020-01-28:
clang-native built successfully.

`
Log data follows:                              
| DEBUG: Executing shell function do_compile                                                   
| ninja: Entering directory /yocto/warrior/build/tmp/work/x86_64-linux/gn-native/79.0.3945.79-r0/chromium-79.0.3945.79/out/Release/gn_build
| [1/182] CXX base/files/file_path_constants.o                                                 
| FAILED: base/files/file_path_constants.o                                                     
| clang++  -MMD -MF base/files/file_path_constants.o.d  -I../../../tools/gn -I. -isystem/yocto/warrior/build/tmp/work/x86_64-linux/gn-native/79.0.3945.79-r0/recipe-sysroot-native/usr/include -O2 -pipe -DNDEBUG -O3 -fdata-sections -ffunction-sections -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -isystem/yocto/warrior/build/tmp/work/x86_64-linux/gn-native/79.0.3945.79-r0/recipe-sysroot-native/usr/include -O2 -pipe -std=c++17 -c ../../../tools/gn/base/files/file_path_constants.cc -o base/files/file_path_constants.o
| In file included from ../../../tools/gn/base/files/file_path_constants.cc:7:                 
| ../../../tools/gn/base/files/file_path.h:109:10: fatal error: 'string_view' file not found                                                                                                  
| #include <string_view>                       
|          ^~~~~~~~~~~~~                       
| 1 error generated.
`
Repeats across all files including <string_view>.

I can see that -std=c++17 is being passed in correctly -- does anything else look out of place?

I am trying to build chromium-ozone-wayland on warrior.

kraj, 2020-01-28:
which branch of meta-clang are you using ? Please try master

jpsalm, 2020-01-28:
I'm on master@7378141606822ef0bb985aaa00e442c9ea806429 (two commits behind)

I will try pulling latest and see if that fixes things.

kraj, 2020-01-28:
I dont think thats going

> I'm on master@7378141606822ef0bb985aaa00e442c9ea806429 (two commits behind)
> 
> I will try pulling latest and see if that fixes things.

that won't help. But it seems that your host is missing the needed headers in C++ runtime
which distro are you on ?

jpsalm, 2020-01-28:
Currently the PC I'm using to build is on ubuntu 16.04 which ships with gcc 5.4.0.

Doesn't yocto create its own native stdlib or does it use the host's?

kraj, 2020-01-28:
@jpsalm no, its host compiler, it does not build runtime libs for host. I think you could either upgrade to newer toolchain on your host or use ubuntu 18.04

something like below will get you gcc9 on your ubuntu install

```
sudo apt-get update -y && 
sudo apt-get upgrade -y && 
sudo apt-get dist-upgrade -y && 
sudo apt-get install build-essential software-properties-common -y && 
sudo add-apt-repository ppa:ubuntu-toolchain-r/test -y && 
sudo apt-get update -y && 
sudo apt-get install gcc-9 g++-9 gcc-9-multilib lib32objc-9-dev lib32stdc++-9-dev lib32gcc-9-dev libobjc-9-dev libstdc++-9-dev libgcc-9-dev  -y && 
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9 60 --slave /usr/bin/g++ g++ /usr/bin/g++-9 && 
sudo update-alternatives --config gcc
```

jpsalm, 2020-01-28:
Thank-you.  You can go ahead and close this.

closed 2020-01-29 by rakuco