OSSystems/meta-browser #354: chromium: update to 80.0.3987.87

msisov, 2020-02-10:
https://chromereleases.googleblog.com/2020/02/stable-channel-update-for-desktop.html

Targets and platforms we need to check:

#### chromium-x11:
- [ ] thud, clang, MACHINE="intel-core2-32"
- [ ] thud, clang, MACHINE="intel-corei7-64"
- [ ] warrior, clang, MACHINE="raspberrypi3"
- [ ] warrior, clang, MACHINE="intel-corei7-64"

#### chromium-ozone-wayland:
- [ ] thud, clang, MACHINE="m3ulcb"
- [ ] thud, clang, MACHINE="qemuarm64"
- [ ] warrior, clang, MACHINE="raspberrypi3"

rakuco, 2020-02-12:
Retitling, 80.0.3987.100 was announced as a stable release yesterday.

rakuco, 2020-02-14:
Retitling again after the new release :-)

rakuco, 2020-02-14:
Though I guess the biggest issue is the M79->M80 upgrade, so if .100 is already on the pipeline we can update to that one first and then update to .106 more quickly later.

msisov, 2020-02-20:
reopened as there is still pending update from .100 -> .106

rakuco, 2020-02-20:
I've retitled the issue, the latest stable is .116 at the moment.

PR #358 (2020-02-26): chromium: update to 80.0.3987.116
Stable release announcements since our latest update:
https://chromereleases.googleblog.com/2020/02/stable-channel-update-for-desktop_18.html

Test-built with:
* chromium-x11
  * thud, MACHINE="intel-corei7-64"
  * thud, MACHINE="intel-core2-32"
  * warrior, MACHINE="intel-corei7-64"
  * warrior, MACHINE="intel-corei7-64"
* chromium-ozone-wayland:
  * thud, MACHINE="qemuarm64"
  * thud, MACHINE="m3ulcb"
  * warrior, MACHINE="raspberrypi3"

Fixes #354
commit d8e4e1cb (2020-02-26): chromium: update to 80.0.3987.116

Stable release announcements since our latest update:
https://chromereleases.googleblog.com/2020/02/stable-channel-update-for-desktop_18.html

Test-built with:
* chromium-x11
  * thud, MACHINE="intel-corei7-64"
  * thud, MACHINE="intel-core2-32"
  * warrior, MACHINE="intel-corei7-64"
  * warrior, MACHINE="intel-corei7-64"
* chromium-ozone-wayland:
  * thud, MACHINE="qemuarm64"
  * thud, MACHINE="m3ulcb"
  * warrior, MACHINE="raspberrypi3"

Fixes #354
closed 2020-02-26 by msisov