OSSystems/meta-browser #357: chromium-x11 soft-crashes frequently with 'Aw, snap'

schnitzeltony, 2020-02-21:
Don't exactly know when it started but it was there with 79.x already. Have decided to wait for 80.x hoping it might go away but it didn't:

On a fresh image chromium (no extensions) frequently displays:

[image]

Reloading helps but the error comes back soon - sometimes by just scrolling a page.

Starting chrome from console I get:

> [3290:7:0221/202546.686655:ERROR:command_buffer_proxy_impl.cc(125)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
> [3290:1:0221/202547.114520:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
> [3290:1:0221/202547.115298:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
> [3361:1:0221/202606.303449:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
> [3361:1:0221/202606.306028:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
> [3361:1:0221/202606.309189:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
> [3361:1:0221/202606.314838:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
> ../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0407

Any ideas?

kraj, 2020-02-21:
this perhaps is related to pread64 syscall or some other not being addressed by libseccomp
is it possible for you to try the same build with glibc 2.30 ( zeus ) and see if it still happens ?

schnitzeltony, 2020-02-21:
@kraj My machine is weak and building 24/7 for other issues. Same answer I sent to you for other issue (was it yesterday?): Will try this weekend - have an extra free monday :grin:

schnitzeltony, 2020-04-20:
Some common comments
1. @kraj I don't have the resources to go back to glibc-2.30 - but I think you are right
2. Nobody else?

I did some digging: The only trace what syscall 0407 might be was found in chromium sources:

sandbox/linux/system_headers/x86_32_linux_syscalls.h:
```c
#define` __NR_clock_nanosleep_time64 407
```
and __this is the only place in chromium code - \_\_NR_clock_nanosleep_time64 is not used anywhere__.

It looks to me as if

* This commit https://sourceware.org/git/?p=glibc.git;a=commit;h=2e44b10b42d68d9887ccab17b76db5d7bbae4fb6 introduced clock_nanosleep_time64 usage in glibc
* chromium's sandbox does not know this syscall and bails out

But I might be on the wrong track...

schnitzeltony, 2020-04-20:
Thought some more - yeah that still  happens from time to time :older_man:

* glibc seems to have replaced time syscalls by time64 syscalls 
* chromium's sandbox is not aware of these (as they seem 32bit-specific?)

Will try a patch extending sandbox handlers by time64-syscalls. Will come back once I know how this works (may take due to chromium build time ...)

kraj, 2020-04-20:
Can you try launching chromium with `--disable-seccomp-filter-sandbox` and see if it works ?
that might be your workaround.  I started to fix it for 64bit time_t but not actively pursuing it due to other priorities see top 3 commits [here](https://github.com/YoeDistro/meta-browser/commits/yoe/mut)

For a proper fix, I think you need to chase down, where the seccomp is being applied, it could
also be a third party dependent library for chromium, see the openssh fix I did upstream so fix
similar [issue](https://github.com/openssh/openssh-portable/pull/161)

schnitzeltony, 2020-04-22:
Fixed by 805b297fc8efe46240f99b2731596a4377bea190

closed 2020-04-22 by schnitzeltony