OSSystems/meta-browser #361: chromium_ozone_wayland does not build any longer with included libdrm

MastaG, 2020-03-10:
Hi I'm building Chromium with it's included exynos drm and gbm instead of the system one.
So I've set:

 use_system_minigbm=false
 use_system_libdrm=false
 use_exynos_minigbm=true

Now it fails to build with:
../../ui/ozone/platform/drm/gpu/drm_display.cc:117:49: error: invalid operands to binary expression ('basic_ostream<char, std::char_traits<char> >' and 'ui::ScopedDrmConnectorPtr' (aka 'unique_ptr<_drmModeConnector, ui::DrmConnectorDeleter>'))
          << " crtc=" << crtc_ << " connector=" << connector_
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~

../../ui/ozone/platform/drm/gpu/drm_display.cc:126:53: error: invalid operands to binary expression ('basic_ostream<char, std::char_traits<char> >' and 'ui::ScopedDrmConnectorPtr' (aka 'unique_ptr<_drmModeConnector, ui::DrmConnectorDeleter>'))
              << " crtc=" << crtc_ << " connector=" << connector_;
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~

msisov, 2020-03-10:
Set ozone_platform_gbm to false.

closed 2020-03-10 by msisov

MastaG, 2020-03-10:
thanks that was it!