OSSystems/meta-browser #362: chromium-ozone-wayland fails to open window on iMX8

Fighter19, 2020-03-12:
I'm on thud, trying to run the latest version of this layer:
chromium-ozone-wayland-80.0.3987.122

The error looks very similar to:
https://github.com/OSSystems/meta-browser/issues/232

<details>
 <summary>Log</summary>

```
root@colibri-imx8x:~# WAYLAND_DEBUG=1 chromium --no-sandbox                                     
[2855686.433]  -> wl_display@1.get_registry(new id wl_registry@2)
[2855686.531]  -> wl_display@1.sync(new id wl_callback@3)
[2855687.061] wl_display@1.delete_id(3)
[2855687.105] wl_registry@2.global(1, "wl_compositor", 4)
[2855687.148]  -> wl_registry@2.bind(1, "wl_compositor", 4, new id [unknown]@4)
[2855687.232] wl_registry@2.global(2, "wl_subcompositor", 1)
[2855687.278]  -> wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@5)
[2855687.334] wl_registry@2.global(3, "wp_viewporter", 1)
[2855687.384] wl_registry@2.global(4, "wp_presentation", 1)
[2855687.429]  -> wl_registry@2.bind(4, "wp_presentation", 1, new id [unknown]@6)
[2855687.484] wl_registry@2.global(5, "zwp_alpha_compositing_v1", 1)
[2855687.527] wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[2855687.571] wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[2855687.608] wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[2855687.644] wl_registry@2.global(9, "wl_data_device_manager", 3)
[2855687.687]  -> wl_registry@2.bind(9, "wl_data_device_manager", 3, new id [unknown]@7)
[2855687.743] wl_registry@2.global(10, "wl_shm", 1)
[2855687.788]  -> wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@8)
[2855687.846] wl_registry@2.global(11, "wl_viv", 1)
[2855687.887] wl_registry@2.global(12, "wl_seat", 5)
[2855687.930]  -> wl_registry@2.bind(12, "wl_seat", 4, new id [unknown]@9)
[2855687.985]  -> wl_data_device_manager@7.get_data_device(new id wl_data_device@10, wl_seat@9)
[2855688.029] wl_registry@2.global(13, "zwp_linux_dmabuf_v1", 3)
[2855688.074]  -> wl_registry@2.bind(13, "zwp_linux_dmabuf_v1", 3, new id [unknown]@11)
[2855688.128]  -> wl_display@1.sync(new id wl_callback@12)
[2855688.162] wl_registry@2.global(14, "zwp_hdr10_metadata_v1", 1)
[2855688.204] wl_registry@2.global(15, "wl_output", 3)
[2855688.248]  -> wl_registry@2.bind(15, "wl_output", 3, new id [unknown]@13)
[2855688.307] wl_registry@2.global(16, "zwp_input_panel_v1", 1)
[2855688.351] wl_registry@2.global(17, "zwp_input_method_v1", 1)
[2855688.388] wl_registry@2.global(18, "zwp_text_input_manager_v1", 1)
[2855688.428]  -> wl_registry@2.bind(18, "zwp_text_input_manager_v1", 1, new id [unknown]@14)
[2855688.483] wl_registry@2.global(19, "zxdg_shell_v6", 1)
[2855688.527]  -> wl_registry@2.bind(19, "zxdg_shell_v6", 1, new id [unknown]@15)
[2855688.577] wl_registry@2.global(20, "wl_shell", 1)
[2855688.613] wl_registry@2.global(21, "weston_desktop_shell", 1)
[2855688.649] wl_registry@2.global(22, "weston_screenshooter", 1)
[2855688.690] wl_callback@3.done(14)
[2855689.333] wl_display@1.delete_id(12)
[2855689.378] wl_seat@9.capabilities(7)
[2855689.403]  -> wl_seat@9.get_pointer(new id wl_pointer@3)
[2855689.444]  -> wl_compositor@4.create_surface(new id wl_surface@16)
[2855689.479]  -> wl_seat@9.get_keyboard(new id wl_keyboard@17)
[2855689.515]  -> wl_seat@9.get_touch(new id wl_touch@18)
[2855689.554] wl_seat@9.name("default")
[2855689.582] zwp_linux_dmabuf_v1@11.modifier(538982482, 100663296, 2)
[2855689.626] zwp_linux_dmabuf_v1@11.modifier(943215175, 100663296, 2)
[2855689.672] zwp_linux_dmabuf_v1@11.modifier(842093144, 100663296, 2)
[2855689.715] zwp_linux_dmabuf_v1@11.modifier(842089048, 100663296, 2)
[2855689.756] zwp_linux_dmabuf_v1@11.modifier(842093121, 100663296, 2)
[2855689.797] zwp_linux_dmabuf_v1@11.modifier(842089025, 100663296, 2)
[2855689.838] zwp_linux_dmabuf_v1@11.modifier(892424792, 100663296, 2)
[2855689.879] zwp_linux_dmabuf_v1@11.modifier(892420696, 100663296, 2)
[2855689.920] zwp_linux_dmabuf_v1@11.modifier(892424769, 100663296, 2)
[2855689.961] zwp_linux_dmabuf_v1@11.modifier(892420673, 100663296, 2)
[2855690.002] zwp_linux_dmabuf_v1@11.modifier(909199186, 100663296, 2)
[2855690.046] zwp_linux_dmabuf_v1@11.modifier(909199170, 100663296, 2)
[2855690.088] zwp_linux_dmabuf_v1@11.modifier(875713112, 100663296, 2)
[2855690.130] zwp_linux_dmabuf_v1@11.modifier(875709016, 100663296, 2)
[... 89 lines cut ...]
#6  0x0000aaaaac958878 in GpuDataManagerImplPrivate () at ../../content/browser/gpu/gpu_data_manager_impl_private.cc:312
#7  0x0000aaaaac957bd0 in make_unique<content::GpuDataManagerImplPrivate, content::GpuDataManagerImpl*> () at /home/patrick/oe-core/build/tmp/work/aarch64-mx8-tdx-linux/chromium-ozone-wayland/80.0.3987.122-r0/recipe-sysroot/usr/lib//aarch64-tdx-linux/8.2.0/../../../include/c++/8.2.0/bits/unique_ptr.h:831
#8  GpuDataManagerImpl () at ../../content/browser/gpu/gpu_data_manager_impl.cc:269
#9  NoDestructor<> () at ../../base/no_destructor.h:54
#10 GetInstance () at ../../content/browser/gpu/gpu_data_manager_impl.cc:19
#11 GetInstance () at ../../content/browser/gpu/gpu_data_manager_impl.cc:14
#12 0x0000aaaaadf6dec8 in ChromeBrowserMainExtraPartsGpu::PostEarlyInitialization() () at ../../chrome/browser/gpu/chrome_browser_main_extra_parts_gpu.cc:28
#13 0x0000aaaaae27fb70 in PostEarlyInitialization () at ../../chrome/browser/chrome_browser_main.cc:852
#14 0x0000aaaaac7a2418 in EarlyInitialization () at ../../content/browser/browser_main_loop.cc:696
#15 0x0000aaaaac7a5f58 in Initialize () at ../../content/browser/browser_main_runner_impl.cc:108
#16 0x0000aaaaac7a1c14 in BrowserMain () at ../../content/browser/browser_main.cc:43
#17 0x0000aaaaadf1cb9c in RunBrowserProcessMain () at ../../content/app/content_main_runner_impl.cc:527
#18 RunServiceManager () at ../../content/app/content_main_runner_impl.cc:960
#19 0x0000aaaaadf1c85c in Run () at ../../content/app/content_main_runner_impl.cc:868
#20 0x0000aaaaadf42840 in Main () at ../../services/service_manager/embedder/main.cc:423
#21 0x0000aaaaadf1b35c in content::ContentMain(content::ContentMainParams const&) () at ../../content/app/content_main.cc:19
#22 0x0000aaaaabcaed8c in ChromeMain () at ../../chrome/app/chrome_main.cc:110
#23 0x0000ffffbe995ce4 in __libc_start_main () from target:/lib/libc.so.6
#24 0x0000aaaaabcaec1c in _start () at ../sysdeps/aarch64/start.S:94
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
</details>

I placed a breakpoint right before the crash and called EnsureInstance using gdb,
which appeared to prevent the crash, but ultimately didn't seem to fix anything.

msisov, 2020-03-12:
Hi.

Your problem is this message - EGL: Warning: No default display support on wayland

Regarding in-process-gpu - it is known to be broken. Hopefully, it will be fixed soon.

Fighter19, 2020-03-12:
I assume I have to do the fixes as described in:
https://github.com/OSSystems/meta-browser/issues/232#issuecomment-472332845

msisov, 2020-03-12:
Hmm.. What is your system specs?

Fighter19, 2020-03-12:
I'm using a Toradex Colibri iMX8QXP board (with Vivante graphics),
with the layers for thud.

The old version of Chromium (71) works on the current image,
although I never properly rebuilt it, and simply transferred it over from NXP's
Reference image.
A newer version (74) still worked in the same environment as the current version. (Although it had some issues).

msisov, 2020-03-12:
AFAIK, it might be a problem with the the wl_display_interface symbol mismatch if chromium is compiled against different library version that the compositor uses.

Make sure you you system libwayland and make sure you have compiled it for your target libraries.

Fighter19, 2020-03-12:
ok, thank you.

I will take that into consideration.
If I understand correctly, it could be that chromium builds using it's
own libwayland which might be incompatible with my system version of wayland?

msisov, 2020-03-12:
Yes.

Fighter19, 2020-03-12:
Ok, thank you for your help.

From what I could gather, libwayland of the system should be used.
As the image is built together with the browser,
I doubt that any mismatch is the cause of the problem.

Instead I think the problem is the implementation of the NXP's libEGL,
which seems to be unable to use to get the
default display managed by wayland.

I'm still wondering however, how it worked with earlier versions of Chromium, then.
I see that there is a wl_display_connect in wayland_connection.cc,
however I don't know how to debug the chromium processes properly.
(It spawns multiple processes and I don't know how to get to the render process, before it fails)

I suspect that InitializeUI in "ozone_platform_wayland.cc",
which should ensure that a valid pointer is passed,
is being executed after InitializeGLOneOffPlatform is called.

However to really test this, I need to wait for the build to finish (once again).

liuming50, 2020-05-11:
@Fighter19 seems you have a workaround by adding "wl_display_connect(0)"? Could you please share the patch maybe, thanks a lot.

I am suffering from this problem on my IMX8QXP board, I tried to upgrade chromium to 81.0.4044.138, but the problem is still there.

Fighter19, 2020-05-11:
I replaced this line:
https://github.com/chromium/chromium/blob/fcea73228632975e052eb90fcf6cd1752d3b42b4/ui/gl/gl_surface_egl.cc#L361
```
return eglGetDisplay (native_display);
```
With this (maybe I've added an include for wl_display on top?):
```
wl_display *display = wl_display_connect(nullptr);
return eglGetDisplay(display);
```
However that's really not much more than a dirty hack.
And should not be taken as a fix. I have no "real" patch.

If I recall correctly, native_display is 0, although it should contain the pointer returned by wl_display_connect.

liuming50, 2020-05-12:
@Fighter19 thanks for the reply, will try that out.

asdasdomar, 2020-05-15:
I'm having the same problem with an imx8mm.

@Fighter19 i tested your changes and it made the browser start but for me it became really slow/unresposive. Can you confirm it's the same for you?

Before 81 i used to start it with --in-process-gpu. Right now the issue when this flag is used is that ui::OzonePlatform::GetInstance() is returning null in gpu::GpuMemoryBufferSupport::IsNativeGpuMemoryBufferConfigurationSupported().

Do any of you know if there's an open issue on the chromium bug tracker for the --in-process-gpu flag?

Thanks

Fighter19, 2020-05-15:
My browser works as fast as usual on the IMX8QXP. (I'm actually quite impressed by how well it runs).

Unfortunately I don't know anything about the Chromium bug tracker.

msisov, 2020-05-15:
> I'm having the same problem with an imx8mm.
>
> @Fighter19 i tested your changes and it made the browser start but for me it became really slow/unresposive. Can you confirm it's the same for you? [...]

--in-process-gpu has been fixed in the upstream.

liuming50, 2020-05-18:
@msisov may I know how it's fixed in upstream, by which commit? Would like to try that as well.

msisov, 2020-05-18:
Please refer to https://bugs.chromium.org/p/chromium/issues/detail?id=1060999

liuming50, 2020-05-26:
@msisov @Fighter19 Just verified chromium-ozone-wayland runs perfectly with '--in-process-gpu' after upgrading to 83.0.4103.61.