OSSystems/meta-browser #365: chromium-ozone-wayland webrtc sound issue.

simonchen007, 2020-03-23:
hello,

There is a bug in chromium-ozone-wayland-79.0.3945.117 , 
Things is that, once the pulseaudio service restarted, the chromium-ozone-wayland  has no audio in webrtc.
the only resoleve way is restart the chromium-ozone-wayland too when  pulseaudio service restarted.  The strange thing is it has audio when play mp3 files without chromium-ozone-wayland restarted, but webrtc has no audio unless restart the chromium-ozone-wayland.

how can I tracing and fix this bug ?

msisov, 2020-03-24:
I'm not a webrtc expert. Please report to upstream. Does it also happen with chromium-x11?

simonchen007, 2020-03-25:
I run chromium-x11 in Debian before, this not happen.

> I'm not a webrtc expert. Please report to upstream. Does it also happen with chromium-x11?

msisov, 2020-03-26:
Can you please report to upstream and share the link here? The meta-browser repo  is not the right place for this kind of things.

simonchen007, 2020-03-26:
I have resolved this issue with another way -- just cancel the pid of pulseaudio ,  it works currently.

msisov, 2020-03-26:
Ok. Thanks for letting us know.

closed 2020-03-26 by msisov