OSSystems/meta-browser #367: chromium-ozone-wayland-80.0.3987.132 missing two includes

MastaG, 2020-03-24:
Hi I'm currently unable to create a PR.

But with the latest master branches of oe-core and meta-clang I need to modify two files in order to successfully build chromium ozone wayland.

third_party/perfetto/include/perfetto/base/task_runner.h: error: ‘uint32_t’ does not name a type
requires <cstdint>

base/trace_event/trace_event_memory_overhead.h:
requires <string>

After these two includes, I can build it 100%.

kraj, 2020-03-24:
I have a patch for clang10 here https://github.com/YoeDistro/meta-browser/commit/1359b9620c8a26c9e6e4fe5226ef8a6752b702f4 give it a shot

MastaG, 2020-03-24:
Thanks I've applied all of them except for the last one:

Applying patch 0001-Remove-glslang-pool_allocator-setAllocator.patch
patching file third_party/glslang/src/glslang/Include/PoolAlloc.h
Hunk #1 FAILED at 304.
1 out of 1 hunk FAILED -- rejects in file third_party/glslang/src/glslang/Include/PoolAlloc.h
Patch 0001-Remove-glslang-pool_allocator-setAllocator.patch can be reverse-applied

I think it's already upstream.

Building now, will report back.

MastaG, 2020-03-24:
all OK :)
Please send a PR, just without the last patch. 

Thanks a lot!

kraj, 2020-03-24:
@MastaG please go ahead with the pull since you have tested it more recently than me and have uptodate content to pull from.

MastaG, 2020-03-25:
Will do, thanks!

closed 2020-03-25 by MastaG