OSSystems/meta-browser #380: chromium: 83 requires python3native?

msisov, 2020-05-29:
With the newest update, I started to get problems with pyhton. Namely, the problem is the following - " No module named pkg_resources ".

Adding to RDEPENDS python-pkg-resources didn't fix the problem.

I tried to start devshell, entered python session and tried to execute the following command -
"from pkg_resources import load_entry_point". That resulted in error. However, doing the same with python3 was fine.

kraj, 2020-05-29:
yes python2 is dropped from OE-core as well. So its a good move.

shr-project, 2020-05-29:
Last time I've checked chromium was far from ready to be built with python3 only, see
https://bugs.chromium.org/p/chromium/issues/detail?id=942720

and
https://bugs.chromium.org/p/chromium/issues/detail?id=61357#c27
> Note that Chromium uses a standalone Python, so distributions removing 2.7 won't be a problem.

rakuco, 2020-05-29:
> > Note that Chromium uses a standalone Python, so distributions removing 2.7 won't be a problem.

That applies to Chromium's own infrastructure (i.e. the way its bots work and how their official binaries are built) as well as when building a git checkout locally.

Downstreams like us using a release tarball will still need Python 2.7 around.

msisov, 2020-06-02:
> The full error caused by using `pythonnative` is
> 
> ```
> python ../../components/schema_org/generate_schema_org_code.py --schema-file ../../third_party/schema_org/schema.jsonld --overrides-file ../../third_party/schema_org/overrides.jsonld --output-dir gen/components/schema_org --templates templates/schema_org_entity_names.cc.tmpl tem
> plates/schema_org_entity_names.h.tmpl templates/schema_org_enums.cc.tmpl templates/schema_org_enums.h.tmpl templates/schema_org_property_configurations.cc.tmpl templates/schema_org_property_configurations.h.tmpl templates/schema_org_property_names.cc.tmpl templates/schema_org_pr
> operty_names.h.tmpl
> Traceback (most recent call last):
>   File "../../components/schema_org/generate_schema_org_code.py", line 22, in <module>
>     env = Environment(loader=PackageLoader('generate_schema_org_code', ''))
>   File "[..]/tmp/work/corei7-64-poky-linux/chromium-x11/83.0.4103.61-r0/chromium-83.0.4103.61/components/schema_org/../../third_party/jinja2/loaders.py", line 222, in __init__
>     from pkg_resources import DefaultProvider, ResourceManager, \
> ImportError: No module named pkg_resources
> ```
> 
> `pkg_resources` comes from setuptools, and in both cases the `python` binary seems to be python2, not python3. Is depending on `python3native` just working around a missing dependency problem?
> 
> > yes python2 is dropped from OE-core as well. So its a good move.
> 
> Was it moved to another layer? The Chromium tree still has a lot of scripts that have not been migrated to Python 3 yet.

it should be setuptools package, but depending on that doesn't resolve the problem

kraj, 2020-06-02:
Yes it’s in meta-python2

rakuco, 2020-06-04:
> it should be setuptools package, but depending on that doesn't resolve the problem

How did you try that out? Adding `setuptools` to the `inherit pythonnative` line in `chromium.inc` seems to have done the trick here.

msisov, 2020-06-05:
Through DEPENDS :facepalm: let me try with inherit

rakuco, 2020-06-11:
Friendly ping. Did adding `inherit setuptools` work for you?

msisov, 2020-06-11:
Yes, it did help. I also found that I used newer clang version for the warrior branch and had to resolve the compilation problem. The bot is doing its final run. This can be closed. I'll send a PR later today or tomorrow. Thanks

rakuco, 2020-06-11:
Thanks for checking. I'll leave this issue open so that you can close it with your PR and have it referenced here automatically.

PR #384 (2020-06-11): chromium: fix warrior build with clang 8 and inherit setuptools.
Switch back to python2 and inherit setuptools instead. Also fixes
warrior build when clang 8 is used (I used newer clang when
tested the recipe before).

fixes #380
closed 2020-06-11 by msisov