OSSystems/meta-browser #381: chromium: consider Dunfell support.

msisov, 2020-06-02:
Dunfell became long term stable in April, 2020. 

Thus, we have 3 stable branches now - Warrior, Zeus and Dunfell. 

Should we support all 3 of them? I'd say we, Igalia, have a need for Warrior for more 2-3 cycles and then can stop using that (not 100%, but most likely).

kraj, 2020-06-02:
I would suggest to drop warrior and keep it as "community supported" and replace it with dunfell. 
Patches still get accepted for warrior, but we should not block the patches etc on warrior.

otavio, 2020-06-02:
I think as could keep warrior being active on demand. So if Igalia needs to apply backports we can accept them but we could speed up the other branches.

kraj, 2020-06-02:
> I think as could keep warrior being active on demand. So if Igalia needs to apply backports we can accept them but we could speed up the other branches.

right, 2.7.4 is the last official warrior release that is happening this week. So its right time to move this to community supported.

rakuco, 2020-06-03:
IIRC, the last time we discussed this was a while ago, in https://github.com/OSSystems/meta-browser/issues/237#issuecomment-474377057, where @otavio suggested supporting master, stable and oldstable (for 2-3 months). I am guessing this means we'd support master (which we never really test but mostly works), dunfell ("current"?), zeus (stable) and warrior (oldstable, for a while)? I'm totally not against not actively supporting warrior, but I think it would be also be a good time to formalize our policies a bit more.

rakuco, 2020-06-03:
One other thing I'm wondering (I can file a separate issue about it if you prefer) is whether we will also treat Dunfell as an LTS release and commit to supporting it for a longer period of time.

msisov, 2020-06-03:
I'd support it as long as it is declared as stable

kraj, 2020-06-03:
> IIRC, the last time we discussed this was a while ago, in [#237 (comment)](https://github.com/OSSystems/meta-browser/issues/237#issuecomment-474377057), where @otavio suggested supporting master, stable and oldstable (for 2-3 months). I am guessing this means we'd support master (which we never really test but mostly works), dunfell ("current"?), zeus (stable) and warrior (oldstable, for a while)? I'm totally not against not actively supporting warrior, but I think it would be also be a good time to formalize our policies a bit more.

now that we have LTS, its perhaps best to stick to yocto release policy for timelines so keep it simple.

msisov, 2020-06-24:
I added dunfell to the test targets. Currently tested targets - warrior, zeus and dunfell.

I think I will keep warrior as a test target until its support level switches to "Community"

closed 2020-06-24 by msisov