OSSystems/meta-browser #388: Chromium X11 build fails on configure step looking for DRI library

dreambeyondorange, 2020-06-15:
Hi,

I am trying to build the chromium-x11 package but it fails to configure.  I am working with dunfell on the master branch of meta-browser.  You can see the depency versions below 

`ndry@u46a4bc29d48753:/portable/jetson-nx/build 12:20:57 $ bitbake chromium-x11
Loading cache: 100% |####################################################################################################| Time: 0:00:00
Loaded 4306 entries from dependency cache.
Parsing recipes: 100% |##################################################################################################| Time: 0:00:00
Parsing of 2854 .bb files complete (2852 cached, 2 parsed). 4306 targets, 127 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-xavier-nx-devkit"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:9c8049068406532c3dd5d8906c595218b0fefd40"
meta-tegra           
contrib              = "dunfell-l4t-r32.4.2:58e4ffb2cc6b50cf10b2c2584e9e053f4fb02e1d"
meta-virtualization  = "dunfell:5c2de3daedd3f65171b9debf938683ce6e7746ea"
meta-browser         = "master:0acb73cd19d7ae3d105f2ae5ef2aa02e8d6c2a35"
meta-rust            = "common-rust-native:3b783652cc9e0e0ff95fdf151eab3e178b5f1667"
meta-clang           = "master:0c9da2c0646005d35b9845b4f91c5b25fb3d43d1"
meta-python2         = "dunfell:e2ef0dd8fa13d6b96e44773b09d07e4817d0a44d"
meta-oe              
meta-networking      
meta-filesystems     
meta-python          
meta-webserver       = "dunfell:cc155826b0b4f202a8fa4ac103b7c38368d32734"
meta-selinux         = "master:f677a09992413f32e6b2c4c607ae44fb42de72c2"

Initialising tasks: 100% |###############################################################################################| Time: 0:00:02
Sstate summary: Wanted 198 Found 191 Missed 7 Current 1639 (96% match, 99% complete)
NOTE: Executing Tasks
ERROR: chromium-x11-83.0.4103.61-r0 do_configure: Execution of '/portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/temp/run.do_configure.7219' failed with exit code 1:
ERROR at //build/config/linux/dri/BUILD.gn:11:20: Script returned non-zero exit code.
  dri_driver_dir = exec_script(pkg_config_script,
                   ^----------
Current dir: /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/chromium-83.0.4103.61/out/Release/
Command: python /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/chromium-83.0.4103.61/build/config/linux/pkg-config.py --dridriverdir dri
Returned 1 and printed out:

Error from pkg-config.

stderr:

Package dri was not found in the pkg-config search path.
Perhaps you should add the directory containing `dri.pc'
to the PKG_CONFIG_PATH environment variable
No package 'dri' found

See //content/gpu/BUILD.gn:134:18: which caused the file to be included.
    configs += [ "//build/config/linux/dri" ]
                 ^-------------------------
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/temp/log.do_configure.7219
Log data follows:
| DEBUG: Executing shell function do_configure
| ERROR at //build/config/linux/dri/BUILD.gn:11:20: Script returned non-zero exit code.
|   dri_driver_dir = exec_script(pkg_config_script,
|                    ^----------
| Current dir: /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/chromium-83.0.4103.61/out/Release/
| Command: python /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/chromium-83.0.4103.61/build/config/linux/pkg-config.py --dridriverdir dri
| Returned 1 and printed out:
| 
| Error from pkg-config.
| 
| stderr:
| 
| Package dri was not found in the pkg-config search path.
| Perhaps you should add the directory containing `dri.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'dri' found
| 
| See //content/gpu/BUILD.gn:134:18: which caused the file to be included.
|     configs += [ "//build/config/linux/dri" ]
|                  ^-------------------------
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/temp/run.do_configure.7219' failed with exit code 1:
| ERROR at //build/config/linux/dri/BUILD.gn:11:20: Script returned non-zero exit code.
|   dri_driver_dir = exec_script(pkg_config_script,
|                    ^----------
| Current dir: /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/chromium-83.0.4103.61/out/Release/
| Command: python /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/chromium-x11/83.0.4103.61-r0/chromium-83.0.4103.61/build/config/linux/pkg-config.py --dridriverdir dri
| Returned 1 and printed out:
| 
| Error from pkg-config.
| 
| stderr:
| 
| Package dri was not found in the pkg-config search path.
| Perhaps you should add the directory containing `dri.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'dri' found
| 
| See //content/gpu/BUILD.gn:134:18: which caused the file to be included.
|     configs += [ "//build/config/linux/dri" ]
|                  ^-------------------------
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/portable/jetson-nx/meta-browser/recipes-browser/chromium/chromium-x11_83.0.4103.61.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4801 tasks of which 4792 didn't need to be rerun and 1 failed.
`

Is there something I am missing here?  I'm not sure how to get the DRI library its looking for.

Thanks!
Ryan

dreambeyondorange, 2020-06-15:
Do I need to turn DRI off or something?  Is it important to building chromium?

dreambeyondorange, 2020-06-16:
If I add mesa as a dependency it succeeds but I don't think that should be required.  What is the proper way to get the DRI support Chromium is expecting?

rakuco, 2020-06-17:
`dri.pc` should be provided by the `virtual/libgl` dependency that we already have. By default, that is mesa, are you getting virtual/libgl from a different recipe?

dreambeyondorange, 2020-06-18:
I believe I am getting libgl from nvidia.  I am working with the NX and the meta-tegra layer, https://layers.openembedded.org/layerindex/branch/master/layer/meta-tegra/  

I am fairly new to developing with Yocto so not sure how to mend these two.  Adding mesa to the DEPENDS variable seemed to work, but I suspect that is the incorrect approach.  

LibGL is provided by the recipe libglvnd from the meta-tegra/recipes-graphics/ but I'm unsure if its being loaded as a dependency for Chromium.

For reference here is my local.conf

CONF_VERSION = "1"
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
MACHINE = "jetson-xavier-nx-devkit"
LICENSE_FLAGS_WHITELIST = "commercial"
DISTRO_FEATURES_append += " virtualization"
IMAGE_INSTALL_append = " cuda-samples tensorrt"
IMAGE_INSTALL_append += "boost ffmpeg"
IMAGE_INSTALL_append += "nvidia-docker"
IMAGE_INSTALL_append += "chromium-x11"
#INHERIT += "rm_work"
GCCVERSION = "8.%"
require contrib/conf/include/gcc-compat.conf
CUDA_VERSION="10.2"
NVIDIA_DEVNET_MIRROR = "file:///portable/nvidia/sdkm_nx"
INIT_MANAGER = "systemd"

and bblayers.conf
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /portable/jetson-nx/meta \
  /portable/jetson-nx/meta-poky \
  /portable/jetson-nx/meta-yocto-bsp \
  /portable/jetson-nx/meta-tegra \
  /portable/jetson-nx/meta-tegra/contrib \
  /portable/jetson-nx/meta-virtualization \
  /portable/jetson-nx/meta-browser \
  /portable/jetson-nx/meta-rust \
  /portable/jetson-nx/meta-clang \
  /portable/jetson-nx/meta-python2 \
  /portable/jetson-nx/meta-openembedded/meta-oe \
  /portable/jetson-nx/meta-openembedded/meta-networking \
  /portable/jetson-nx/meta-openembedded/meta-filesystems \
  /portable/jetson-nx/meta-openembedded/meta-python \
  /portable/jetson-nx/meta-openembedded/meta-webserver \
  /portable/jetson-nx/meta-selinux \
  "
Thanks for your help!