OSSystems/meta-browser #389: firefox build fails mach cannot find python

dreambeyondorange, 2020-06-15:
I am trying to build with stock repos from dunfell and the master branch of meta-browser.  Firefox fails to build because some internal tool cannot find python.  I opened the dev shell and python is available so not sure why or what this error is about.

ndry@u46a4bc29d48753:/portable/jetson-nx 16:04:27 $ bitbake firefox
Loading cache: 100% |####################################################################################################| Time: 0:00:00
Loaded 4306 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-xavier-nx-devkit"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:9c8049068406532c3dd5d8906c595218b0fefd40"
meta-tegra           
contrib              = "dunfell-l4t-r32.4.2:58e4ffb2cc6b50cf10b2c2584e9e053f4fb02e1d"
meta-virtualization  = "dunfell:5c2de3daedd3f65171b9debf938683ce6e7746ea"
meta-browser         = "master:0acb73cd19d7ae3d105f2ae5ef2aa02e8d6c2a35"
meta-rust            = "common-rust-native:3b783652cc9e0e0ff95fdf151eab3e178b5f1667"
meta-clang           = "master:0c9da2c0646005d35b9845b4f91c5b25fb3d43d1"
meta-python2         = "dunfell:e2ef0dd8fa13d6b96e44773b09d07e4817d0a44d"
meta-oe              
meta-networking      
meta-filesystems     
meta-python          
meta-webserver       = "dunfell:cc155826b0b4f202a8fa4ac103b7c38368d32734"
meta-selinux         = "master:f677a09992413f32e6b2c4c607ae44fb42de72c2"

Initialising tasks: 100% |###############################################################################################| Time: 0:00:01
Sstate summary: Wanted 190 Found 185 Missed 5 Current 1421 (97% match, 99% complete)
NOTE: Executing Tasks
ERROR: firefox-68.9.0esr-r0 do_configure: Execution of '/portable/jetson-nx/build/tmp/work/aarch64-poky-linux/firefox/68.9.0esr-r0/temp/run.do_configure.31788' failed with exit code 127:
./mach: 9: exec: python: not found
WARNING: exit code 127 from a shell command.

ERROR: Logfile of failure stored in: /portable/jetson-nx/build/tmp/work/aarch64-poky-linux/firefox/68.9.0esr-r0/temp/log.do_configure.31788
Log data follows:
| DEBUG: Executing shell function do_configure
| ./mach: 9: exec: python: not found
| WARNING: exit code 127 from a shell command.
| ERROR: Execution of '/portable/jetson-nx/build/tmp/work/aarch64-poky-linux/firefox/68.9.0esr-r0/temp/run.do_configure.31788' failed with exit code 127:
| ./mach: 9: exec: python: not found
| WARNING: exit code 127 from a shell command.
| 
ERROR: Task (/portable/jetson-nx/meta-browser/dynamic-layers/rust-layer/recipes-browser/firefox/firefox_68.9.0esr.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3802 tasks of which 3801 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /portable/jetson-nx/meta-browser/dynamic-layers/rust-layer/recipes-browser/firefox/firefox_68.9.0esr.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

I have no idea how to debug this.

shr-project, 2020-06-15:
See https://github.com/OSSystems/meta-browser/issues/338 with possible work around.

dreambeyondorange, 2020-06-15:
Its unclear to me after reading that post what the work around to implement actually is

rcla, 2020-07-03:
In my case, I fixed it by adding the following line:
`HOSTTOOLS += "python python2.7"`

In the File:
meta-browser/conf/layer.conf

This would be:
[https://github.com/shr-project/meta-browser/blob/bc2d0e3dcc1175f7bfc8bd7f7ed60953c7553635/conf/layer.conf](https://github.com/shr-project/meta-browser/blob/bc2d0e3dcc1175f7bfc8bd7f7ed60953c7553635/conf/layer.conf#L12)

outbackdingo, 2020-07-13:
why python2.7 its known to be DEAD..... is it not possible to simply build a browser firefox or other without issues?

kraj, 2020-07-13:
> why python2.7 its known to be DEAD..... is it not possible to simply build a browser firefox or other without issues?
yes desire is to remove dependence, however, firefox build system is heavily entrenched in it and there is work happening in Mozilla community for a few years see [this](https://wiki.mozilla.org/Firefox/Python_3_Migration)