OSSystems/meta-browser #401: chromium 84.0.4147.105 requires xcb-proto

msisov, 2020-08-10:
Hi, Chromium is going away from xlib and adopts libxcb instead.

However, there is one problem. We are still using python2 as chromium is not fully compatible with python3. And there is one package missing from python2 - xcbgen.

Basically, we need xtypes.py script for that. Here is the error log

```
FAILED: gen/ui/gfx/x/bigreq_undef.h gen/ui/gfx/x/bigreq.h gen/ui/gfx/x/bigreq.cc 
python ../../ui/gfx/x/gen_xproto.py /usr/share/xcb/bigreq.xml gen/ui/gfx/x/bigreq_undef.h gen/ui/gfx/x/bigreq.h gen/ui/gfx/x/bigreq.cc
Traceback (most recent call last):
  File "../../ui/gfx/x/gen_xproto.py", line 1028, in <module>
    sys.exit(main())
  File "../../ui/gfx/x/gen_xproto.py", line 1018, in main
    import xcbgen.xtypes

```

Looking at the sysroot, I could only find the xcbgen package with the required .py file in python3. In contrast, my ubuntu 20 didn't have that package as well, because its support seemed to be discontinued (https://packages.ubuntu.com/search?keywords=xcbgen&searchon=names).  

`./usr/lib/python3.8/site-packages/xcbgen/xtypes.py`

Any hints?

msisov, 2020-08-10:
@rakuco explicit \cc

otavio, 2020-08-10:
Did you check if the current release, of `xcbgen`, also builds on py2?

msisov, 2020-08-10:
What do you mean by "the current release" ? Our friends at Google just recently started migration to xcb from xlib. And it's the first release (m84) that requires xcbgen

otavio, 2020-08-10:
I mean `python-xcbgen`. If the current release also works on Python 2.

If it does, we can have a recipe for it in a `recipes-staging/python/` so it is a workaround for now.

msisov, 2020-08-10:
Actually, bitbake says

`xcb-proto RPROVIDES python-xcbgen`

I do have xcb-proto in DEPENDS in the chromium recipe. And still I can't see the xcbgen in the sysroot's python2 folder.

msisov, 2020-08-11:
so, it's the xcb-proto that's responsible for that.

Forcing it to work with python2 results in installing xcbgen to python2.

What would be the correct thing to do here? Adding a .bbappend file for xcb-proto to the meta-browser recipe?

```
diff --git a/meta/recipes-graphics/xorg-proto/xcb-proto_1.13.bb b/meta/recipes-graphics/xorg-proto/xcb-proto_1.13.bb
index 7467090920..f5e0aa4e13 100644
--- a/meta/recipes-graphics/xorg-proto/xcb-proto_1.13.bb
+++ b/meta/recipes-graphics/xorg-proto/xcb-proto_1.13.bb
@@ -15,7 +15,7 @@ SRC_URI = "http://xcb.freedesktop.org/dist/${BP}.tar.bz2"
 SRC_URI[md5sum] = "abe9aa4886138150bbc04ae4f29b90e3"
 SRC_URI[sha256sum] = "7b98721e669be80284e9bbfeab02d2d0d54cd11172b72271e47a2fe875e2bde1"
 
-inherit autotools pkgconfig python3native
+inherit autotools pkgconfig pythonnative
 
 PACKAGES += "python-xcbgen"
 

```

otavio, 2020-08-11:
We have a hard problem. We should never change core recipes especially if we're changing how they work for everyone. 

How far is Chromium from work with Python 3?

msisov, 2020-08-11:
I think it's very far away - https://bugs.chromium.org/p/chromium/issues/detail?id=942720

There are plenty of open bugs. @rakuco should know better as he is the one who sends patches to that part of chromium.

otavio, 2020-08-11:
Oh my; and how difficult would be to revert the xcb change?

msisov, 2020-08-11:
It looks like python3 is used since xcb-proto 1.13 - http://cgit.openembedded.org/openembedded-core/commit/meta/recipes-graphics/xorg-proto/xcb-proto_1.13.bb?h=dunfell&id=f5341f043ed63db717c74677ff831fd5de7ce7ef

Warrior and dunfell use that version.

otavio, 2020-08-11:
No, I meant to revert Chromium to use xlib ?

otavio, 2020-08-11:
Or make this explicit module to use python3, during build.

msisov, 2020-08-12:
Nope, it's going to be a non-trivial task to revert that. There are tons of changes (I was the one who reviewed a few of them as well).

msisov, 2020-08-12:
Actually, I missed some patches. Chromium ships own xcbproto now. https://chromium-review.googlesource.com/c/chromium/src/+/2267923

closed 2020-08-12 by msisov