OSSystems/meta-browser #415: chromium-x11 85 failing to compile on machines without GBM

ekapllaj, 2020-10-06:
Hi,
I'm trying to compile chromium 85.0.4183.102 for I.MX6 machine and it is failing on configuration step. On these machines, there is no support for GBM.

`| Returned 1. 
| stderr: 
|  
| Package gbm was not found in the pkg-config search path. 
| Perhaps you should add the directory containing `gbm.pc' 
| to the PKG_CONFIG_PATH environment variable 
| No package 'gbm' found 
| Could not run pkg-config. 
|  
| See //third_party/minigbm/BUILD.gn:122:3: whence it was called. 
|   pkg_config("libgbm") { 
|   ^--------------------- 
| See //ui/gfx/linux/BUILD.gn:41:5: which caused the file to be included. 
|     "//third_party/minigbm", 
|     ^---------------------- 
| WARNING: exit code 1 from a shell command.`

This problem is present from version 81.0.4044.122 (commit a48be65f493dbf5b9eb35ae3918a2de0e59e0ebe).

On previous versions of chrome, we could overcome this problem by using variable "use_system_minigbm=false", but there seems to be a problem now. Chromium compiles minigbm, but fails to compile gpu_memory_buffer_support_x11 target:

`| In file included from ../../ui/gfx/linux/gpu_memory_buffer_support_x11.cc:21: 
| ../../ui/gfx/linux/gbm_device.h:8:10: fatal error: 'gbm.h' file not found 
| #include <gbm.h> 
|          ^~~~~~~ 
| 1 error generated. `

To make this work, I had to make a small patch to add "//third_party/minigbm/src" to the include path because it couldn't find the gbm.h file provided by minigbm:

`diff --git a/ui/gfx/linux/BUILD.gn b/ui/gfx/linux/BUILD.gn
index c54b85f..b8c874d 100644
--- a/ui/gfx/linux/BUILD.gn
+++ b/ui/gfx/linux/BUILD.gn
@@ -61,6 +61,8 @@ if (use_x11) {
       "//ui/gfx/x",
     ]
     defines = [ "IS_GBM_SUPPORT_X11_IMPL" ]
+
+    include_dirs = [ "//third_party/minigbm/src" ]
   }
 }`

I think a condition should be added on this file in order to use the built-in minigbm headers if use_system_minigbm variable is set to false.

I tried to do something like this:
`if(!use_system_minigbm){
  include_dirs = [ "//third_party/minigbm/src" ]
}
`

but i got an error:
`| ERROR at //ui/gfx/linux/BUILD.gn:65:10: Undefined identifier
|     if (!use_system_minigbm) {
|          ^-----------------
`

I'm not so experienced on chromium build system, but i think this variable is not visible on this BUILD.gn file. So far I'm using the patch limited on mx6 machines, but I think it would be better if this issue is fixed upstream.

rakuco, 2020-10-06:
@msisov even though this is chromium-x11, is this issue something you have experience with upstream?

msisov, 2020-10-06:
in your gn args, switch use_system_minigbm to false. By default, chrome tries to use system libgbm.

ekapllaj, 2020-10-06:
Hello, and thanks for the quick reply!!
I actually have that variable set on my gn args, and minigbm is being compiled, the problem I see here is:
even setting use_system_minigbm to false, chrome fails to compile gpu_memory_buffer_support_x11 because it doesn't consider that condition.

We should condition gpu_memory_buffer_support_x11 target to use minigbm when use_system_minigbm=false, and this is currently not the case.

I think the best thing to do  could be to patch /ui/gfx/linux/BUILD.gn file in order to consider the use_system_minigbm value:
something like this:
```
if(!use_system_minigbm){
    include_dirs = [ "//third_party/minigbm/src" ]
}
```

I mean, if we are not using system gbm, then we should include //third_party/minigbm/src into the path in order to avoid failure on non gbm systems.

msisov, 2020-10-06:
The logic whether chrome must use system minigbm or the one from the third_party is located at //third_party/minigbm/BUILD.gn. Everything is correct there.

Doing "include_dirs = [ "//third_party/minigbm/src" ]" is incorrect.

ekapllaj, 2020-10-06:
Yes you are right about //third_party/minigbm/BUILD.gn, and I also think setting use_system_minigbm=false should be enough, but currently it is not. 
The problem is on file /ui/gfx/linux/BUILD.gn, which adds:
```
if (use_x11) {
  component("gpu_memory_buffer_support_x11") {
    sources = [
      "gpu_memory_buffer_support_x11.cc",
      "gpu_memory_buffer_support_x11.h",
    ]
    deps = [
      ":drm",
      ":gbm",
      "//base",
      "//skia",
      "//ui/gfx:buffer_types",
      "//ui/gfx:memory_buffer",
      "//ui/gfx/x",
    ]
    defines = [ "IS_GBM_SUPPORT_X11_IMPL" ]
  }
}
```

So it ads the gbm dependency, but when compiling gpu_memory_buffer_support_x11 target, it cannot find gbm.h because it doesn't have "//third_party/minigbm/src" on the compiler FLAGS. I don't know if my solution is correct or not, but somehow we should inform the compiler to look under //third_party/minigbm/src/ for gbm.h when use_system_minigbm=false, and this is currently missing.

Setting just "use_system_minigbm=false" it fails with:
```
| arm-poky-linux-gnueabi-clang++   -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/elvis/work/Zeus/hmi5x/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot -MMD -MF obj/ui/gfx/linux/gpu_memory_buffer_support_x11/gpu_memory_buffer_support_x11.o.d -DIS_GBM_SUPPORT_X11_IMPL -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_40 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_40 -DWEBP_EXTERN=extern -DSK_CODEC_DECODES_PNG -DSK_CODEC_DECODES_WEBP -DSK_ENCODE_PNG -DSK_ENCODE_WEBP -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_GL -DSK_CODEC_DECODES_JPEG -DSK_ENCODE_JPEG -DSK_USE_LIBGIFCODEC -DSK_VULKAN_HEADER=\"../../skia/config/SkVulkanConfig.h\" -DSK_VULKAN=1 -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DVK_USE_PLATFORM_XCB_KHR 
-I../.. -Igen -I../../third_party/perfetto/include -Igen/third_party/perfetto/build_config -Igen/third_party/perfetto 
-I../../third_party/libwebp/src -I../../third_party/boringssl/src/include -I../../third_party/protobuf/src 
-Igen/protoc_out -I../../third_party/skia -I../../third_party/libgifcodec -I../../third_party/vulkan/include 
-I../../third_party/vulkan_headers/include -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -Xclang -fdebug-compilation-dir -Xclang . -mfpu=neon -mthumb -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare 
-I/home/elvis/work/Zeus/hmi5x/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot/usr/include/glib-2.0 
-I/home/elvis/work/Zeus/hmi5x/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot/usr/lib/glib-2.0/include 
-std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=/home/elvis/work/Zeus/hmi5x/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/home/elvis/work/Zeus/hmi5x/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/home/elvis/work/Zeus/hmi5x/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/elvis/work/Zeus/hmi5x/build//tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot-native= -g1  -c ../../ui/gfx/linux/gpu_memory_buffer_support_x11.cc -o obj/ui/gfx/linux/gpu_memory_buffer_support_x11/gpu_memory_buffer_support_x11.o 

| In file included from ../../ui/gfx/linux/gpu_memory_buffer_support_x11.cc:21: 

| ../../ui/gfx/linux/gbm_device.h:8:10: fatal error: 'gbm.h' file not found 

| #include <gbm.h> 

|          ^~~~~~~ 

| 1 error generated. 
```

msisov, 2020-10-06:
Please check if the "gbm" target in the at the //ui/gfx/linux/BUILD.gn has //third_party/minigbm in deps like here [1]

[1] https://source.chromium.org/chromium/chromium/src/+/master:ui/gfx/linux/BUILD.gn;l=41

msisov, 2020-10-06:
There is include_dirs in the minigbm - https://source.chromium.org/chromium/chromium/src/+/master:third_party/minigbm/BUILD.gn;l=32;bpv=1

ekapllaj, 2020-10-06:
Yes I do have the same //ui/gfx/linux/BUILD.gn file (the only difference is ozone_platform_gbm has changed to ozone_platform_drm on the assert).

I also see the include_dirs in the minigbm BUILD.gn, I don't know yet why this information is not propagating correctly. I removed my patch and started a build with only the "use_system_minigbm=false" to replicate the problem and debug a little further.

msisov, 2020-10-06:
do you have any other changes that may override chromium build config?

ekapllaj, 2020-10-06:
I was just looking on the config file, this is the do_configure log:

```
DEBUG: Executing shell function do_configure
WARNING at the command-line "--args":1:1214: Build argument has no effect.

use_cups=false          
ffmpeg_branding="Chromium" 
proprietary_codecs=false          
is_component_build=false         
use_gnome_keyring=false         
use_jumbo_build=false         
use_kerberos=false         
use_pulseaudio=false         
use_system_freetype=true         
use_system_libjpeg=true  
enable_js_type_check=false 
host_pkg_config="pkg-config-native" 
is_debug=false 
is_official_build=true 
use_custom_libcxx=false 
use_lld=false 
use_gold=true 
symbol_level=0 
enable_remoting=false 
enable_nacl=false 
use_sysroot=false 
treat_warnings_as_errors=false 
is_cfi=false 
fieldtrial_testing_like_official_build=true          
google_api_key="invalid-api-key"         
google_default_client_id="invalid-client-id"         
google_default_client_secret="invalid-client-secret"           
custom_toolchain="//build/toolchain/yocto:yocto_target"        
gold_path=""         
host_toolchain="//build/toolchain/yocto:yocto_native"         
is_clang=true         
clang_base_path="/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot-native/usr/bin"         
clang_use_chrome_plugins=false         
linux_use_bundled_binutils=false         
target_cpu="arm"         
v8_snapshot_toolchain="//build/toolchain/yocto:yocto_target"  
use_lld=true use_gold=false         
arm_float_abi="hard"         
arm_fpu="neon"         
arm_tune="cortex-a9"        
arm_version=7  
use_system_minigbm=false
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             ^----
The variable "linux_use_bundled_binutils" was set as a build argument
but never appeared in a declare_args() block in any buildfile.

To view all possible args, run "gn args --list <out_dir>"

The build continued as if that argument was unspecified.

Done. Made 12694 targets from 2280 files in 27863ms
DEBUG: Shell function do_configure finished
DEBUG: Executing python function do_qa_configure
DEBUG: Python function do_qa_configure finished 
```

msisov, 2020-10-06:
What yocto config do you have? Can you paste it here so that I can replicate it locally.

ekapllaj, 2020-10-06:
I'm using Yocto Zeus, for i.mx6 target. The only change i make to chrome is adding this patch:
```
# gbm is availiable only for mx8
GN_ARGS_append_mx6 = " use_system_minigbm=false "
```

I actually had the doubt I may have some strange patch creating problems, so I tested also the official BSP from freescale, which also failed on the same way.

To replicate the problem, just download the BSP:
repo init -u git://source.codeaurora.org/external/imx/imx-manifest.git -b imx-linux-zeus -m imx-5.4.47-2.2.0.xml
repo sync

DISTRO=fsl-imx-x11 MACHINE=imx6qsabresd source imx-setup-release.sh -b build
bitbake chromium-x11

This will fail because there is no provider for gbm. If you set GN_ARGS_append_mx6 = " use_system_minigbm=false " on the bbappend, then you can replicate the problem.

ekapllaj, 2020-10-06:
Now I'm mostly there (it fails at around 35% of compile process), it has already compiled the minigbm, and on this case, the path is correctly set ( -I../../third_party/minigbm/src ):

```
[5266/44743] arm-poky-linux-gnueabi-clang   -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-secu
    rity -Werror=format-security --sysroot=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot -MMD -MF obj/third_party/minigbm/minigbm/    gbm.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_
    ENABLED=0 
-I../.. -Igen 

-I../../third_party/minigbm/src 

-fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables    -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -Xclang -fdebug-compilation-dir -Xclang . -mfpu=neon -mthumb -O2
    -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-mislea
    ding-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-implicit-int-flo
    at-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -I/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-ne
    on-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot/usr/include/libdrm -std=c11 -Wno-implicit-fallthrough  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/build_ase    m-hmi5x-mx6/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work
    /cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot=                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gn
    ueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot-native= -g1 -c ../../third_party/minigbm/src/gbm.c -o obj/third_party/minigbm/minigbm/gbm.o   
```

ekapllaj, 2020-10-06:
This is all i can find on the compile log about gbm:
```
[5258/44743] arm-poky-linux-gnueabi-clang   -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot -MMD -MF obj/third_party/minigbm/minigbm/dri.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -I../../third_party/minigbm/src -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -Xclang -fdebug-compilation-dir -Xclang . -mfpu=neon -mthumb -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -I/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot/usr/include/libdrm -std=c11 -Wno-implicit-fallthrough  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot=                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot-native= -g1 -c ../../third_party/minigbm/src/dri.c -o obj/third_party/minigbm/minigbm/dri.o
[5259/44743] arm-poky-linux-gnueabi-clang   -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot -MMD -MF obj/third_party/lzma_sdk/lzma_sdk/LzmaDec.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -D_7ZIP_ST -D_7Z_NO_METHODS_FILTERS -D_LZMA_PROB32 -I../.. -Igen -I../../third_party/lzma_sdk -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -Xclang -fdebug-compilation-dir -Xclang . -mfpu=neon -mthumb -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -Wno-self-assign -std=c11 -Wno-implicit-fallthrough  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0=/usr/src/debug/chromium-x11/85.0.4183.102-r0                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe-sysroot=                      -fdebug-prefix-map=/mnt/DEVEL/Zeus/mx6/hmi-x11/build/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/chromium-x11/85.0.4183.102-r0/recipe [... cut ...] _x11.o
In file included from ../../ui/gfx/linux/gpu_memory_buffer_support_x11.cc:21:
../../ui/gfx/linux/gbm_device.h:8:10: fatal error: 'gbm.h' file not found
#include <gbm.h>
         ^~~~~~~
1 error generated.

```

When compiling targets defined on "third_party/minigbm/BUILD.gn, it has correctly set `-I../../third_party/minigbm/src` and all goes well, when compiling targets defined on  //ui/gfx/linux/BUILD.g instead, it doesn't have `-I../../third_party/minigbm/src` and it fails to find gbm.h

rakuco, 2020-10-06:
> When compiling targets defined on "third_party/minigbm/BUILD.gn, it has correctly set `-I../../third_party/minigbm/src` and all goes well, when compiling targets defined on //ui/gfx/linux/BUILD.g instead, it doesn't have `-I../../third_party/minigbm/src` and it fails to find gbm.h

For the record, I can reproduce this with today's upstream master branch by not having `gbm.h` installed system-wide and building with `use_system_minigbm=false` and `use_sysroot=false` (the latter is important, otherwise the build will use `gbm.h` from the Debian sysroot). @msisov was this supposed to work? I don't even know what GBM really is, so I'm tempted to leave this in your hands :-)

msisov, 2020-10-06:
It's the libgbm library - https://packages.debian.org/sid/libgbm-dev

I'll report to upstream.

msisov, 2020-10-07:
I was said it still has been assumed that it was expected that libgbm was installed.

To fix the problem, something like this should be done (check if minigbm is used and include the header appropriately) - see https://source.chromium.org/chromium/chromium/src/+/master:third_party/iccjpeg/iccjpeg.h;l=25;drc=a1a8185c72545ce458e3898e98c0dae095977f0e

ekapllaj, 2020-10-07:
I was reading the documentation of GN, and came across this example, which may explain why the //third_party/minigbm/src/ is not visible for gpu_memory_buffer_support_x11 target.

```
  # This target can include files from "c" but not from
  # "super_secret_implementation_details".
  executable("a") {
    deps = [ ":b" ]
  }

  shared_library("b") {
    deps = [ ":super_secret_implementation_details" ]
    public_deps = [ ":c" ]
  }
```

On file /ui/gfx/linux/BUILD.gn we declare gpu_memory_buffer_support_x11 target, which deps on gbm, which deps on //third_party/minigbm/. And since this last dependency is not a `public_deps`, than the target cannot include the headers from minigbm folder.

zpfvo, 2021-04-13:
Is there any news on this problem? I try to build chromium-ozone-wayland on disto fslc-xwayland  (imx6 dunfell) and I get the exact same error:
```
NOTE: recipe chromium-ozone-wayland-80.0.3987.132-r0: task do_configure: Started
ERROR: chromium-ozone-wayland-80.0.3987.132-r0 do_configure: Execution of '.../build_xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/80.0.3987.132-r0/temp/run.do_configure.17629' failed with exit code 1:
ERROR at .../build/config/linux/pkg_config.gni:103:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: .../build_xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/80.0.3987.132-r0/chromium-80.0.3987.132/out/Release/
Command: python .../build_xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/80.0.3987.132-r0/chromium-80.0.3987.132/build/config/linux/pkg-config.py gbm
Returned 1.
stderr:

Package gbm was not found in the pkg-config search path.
Perhaps you should add the directory containing `gbm.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gbm' found
Could not run pkg-config.

See //third_party/minigbm/BUILD.gn:121:3: whence it was called.
  pkg_config("libgbm") {
  ^---------------------
See //media/gpu/test/BUILD.gn:172:7: which caused the file to be included.
      "//third_party/minigbm",
      ^----------------------
WARNING: exit code 1 from a shell command.
```

I also tried a few different versions (80 - 89) of chromium, but the error persists.

msisov, 2021-04-13:
I believe nobody worked on that. Patches are welcomed :)

xtrinch, 2022-06-18:
Same error here.

bmx666, 2022-07-14:
could you replace it on macro which check dependency and people doesn't need to reinvent wheel over and over?

`"${@bb.utils.contains('DEPENDS', 'virtual/libgbm', 'use_wayland_gbm=true', 'use_wayland_gbm=false', d)}"`