OSSystems/meta-browser #418: Adding/buildin chrome extension

nandra, 2020-10-21:
Hi,

I'm using chromium from this meta layer and all seems to be fine. I plan to add some extension but automatic install fails due reasons (root user, guest account, various restrictions, etc.. Is there any way how to easily add extension to compile chrome? Or I need to do special recipe which will fetch extension and somehow install to proper path for chromium to be able to see it? Thanks for any info.

r1mikey, 2020-10-21:
What we do is use a separate recipe to install our extensions.

The crx file and associate $HASH.json file simply need to be installed to `${D}/usr/share/chromium/extensions`.

nandra, 2020-10-21:
@r1mikey thanks a lot. I'll give a try. :+1:

nandra, 2020-10-23:
Thanks that did the trick. Closing.

closed 2020-10-23 by nandra