OSSystems/meta-browser #420: Consider supporting "zeus" again

rakuco, 2020-10-29:
The change to `LAYERSERIES_COMPAT_browser-layer` in #412 were not properly discussed. We are still interested in supporting the "zeus" branch.

otavio, 2020-10-29:
@rakuco, the biggest problem is maintaining old metadata and add workarounds for incompatibilities. Using specific branches seems better, and backporting the new releases is not a problem if someone maintains it.

r1mikey, 2020-10-29:
Context here: in discussing this, the right place to fix the 25 character limit would be within layerindex, not in the numerous layers that support a large number of Yocto versions via LAYERSERIES_COMPAT.

I don't think that anybody disagrees that maintaining compatible metadata can be very difficult (and sometimes a branch could be the easier or more correct option), but it doesn't make sense that the fix for "change the width of a database field" is to require modification of an unbounded number of YP compatible layers from around the Internet.

otavio, 2020-10-29:
Have the layer index issue to be reported?

r1mikey, 2020-10-29:
> Have the layer index issue to be reported?

That's a great question.  We've just had this discussion on #419, so I'd assume "not yet".  There doesn't seem to be an issue tracker for layerindex, and I'm loath to add to my existing mountain of email by joining the OE mailing list just to report one issue.

shr-project, 2020-10-29:
How would you conditionally add meta-python2 to LAYERDEPENDS? As it shouldn't be used for zeus builds, but it's mandatory for dunfell and gatesgarth?

msisov, 2020-10-30:
> The change to `LAYERSERIES_COMPAT_browser-layer` in #412 were not properly discussed. We are still interested in supporting the "zeus" branch.

Though, I'm also fine with dropping zeus at all and testing for Gatesgarth and Dunfell.

r1mikey, 2020-10-30:
I'm OK with dropping Zeus.

otavio, 2020-10-30:
Let us help customers to move on and move to supported releases; Zeus for me is dead for new projects except if explicitly required (yes, I know this happens), but in this case, the customer needs to be aware he is using an old release and new stuff is not supported or require extra work.

msisov, 2020-11-03:
Ok. Then, I'm dropping Zeus from now on. Thank you for this discussion.

closed 2020-11-03 by msisov