OSSystems/meta-browser #423: Context menu crashes Chromium on i.MX8

daescalona, 2020-11-16:
Every time I try to open context menu or application menu in Chromium the application crashes and leaves the display unusable. This is the error that is printed:

`[981:1003:1116/115148.417683:ERROR:shared_image_manager.cc(268)] SharedImageManager::ProduceOverlay: Trying to produce a Overlay representation from an incompatible mailbox.
[981:1003:1116/115148.419556:ERROR:shared_image_manager.cc(268)] SharedImageManager::ProduceOverlay: Trying to produce a Overlay representation from an incompatible mailbox.
[981:1003:1116/115148.421400:ERROR:shared_image_manager.cc(268)] SharedImageManager::ProduceOverlay: Trying to produce a Overlay representation from an incompatible mailbox.
[981:1003:1116/115153.845384:ERROR:shared_image_manager.cc(268)] SharedImageManager::ProduceOverlay: Trying to produce a Overlay representation from an incompatible mailbox.
[981:1003:1116/115153.846195:ERROR:shared_image_manager.cc(268)] SharedImageManager::ProduceOverlay: Trying to produce a Overlay representation from an incompatible mailbox.
[981:1003:1116/115153.847110:ERROR:shared_image_manager.cc(268)] SharedImageManager::ProduceOverlay: Trying to produce a Overlay representation from an incompatible mailbox.`

I am using Chromium v85.0.4183.102 running on a i.MX8 device.

ChrisGRSEB, 2020-12-22:
Hi,
According to you is it the same issue that I mentionned here : https://github.com/OSSystems/meta-browser/issues/427  ?
Do you find a solution to resolve this issue ?

Best regards,

msisov, 2021-01-04:
Hi. The development of Wayland support is still going. I'd suggest to use the latest available Chromium version rather than searching for a patch to backport.