OSSystems/meta-browser #424: Kiosk/Fullscreen mode in Chromium fails on i.MX8

daescalona, 2020-11-16:
Using Chromium v85.0.4183.102 in a i.MX8 I am unable to get it to work on Kiosk or Fullscreen mode. Every time I try that this error is printed forever until I stop the application:

`[     1] ioctl(DRM_IOCTL_GEM_CLOSE) failed
[817:838:1116/114711.361451:ERROR:gbm_pixmap_wayland.cc(58)] Cannot create bo with format= BGRA_8888 and usage SCANOUT
[817:838:1116/114711.361511:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(218)] Failed to create pixmap 1280x800, BGRA_8888, usage SCANOUT
[     2] ioctl(DRM_IOCTL_GEM_CLOSE) failed
[817:838:1116/114711.361888:ERROR:gbm_pixmap_wayland.cc(58)] Cannot create bo with format= BGRA_8888 and usage GPU_READ
[817:838:1116/114711.361936:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(218)] Failed to create pixmap 1280x800, BGRA_8888, usage GPU_READ
[817:838:1116/114711.361970:ERROR:shared_image_backing_factory_gl_texture.cc(1437)] CreateSharedImage: Failed to create bindable image
[817:838:1116/114711.361994:ERROR:shared_image_factory.cc(491)] CreateSharedImage: could not create backing.
[817:838:1116/114711.367047:ERROR:skia_output_surface_impl_on_gpu.cc(1701)] Failed to make current.
[817:838:1116/114711.367310:ERROR:skia_output_surface_impl_on_gpu.cc(1701)] Failed to make current.
[863:863:1116/114711.371576:ERROR:command_buffer_proxy_impl.cc(122)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
[     3] ioctl(DRM_IOCTL_GEM_CLOSE) failed
`

I remember this working on previous versions. Does anyone have the same issue?

Thanks in advance.

msisov, 2020-11-16:
afaik, imx8 doesn't support drm render nodes. secondly, chromium should fallback to sw compositing instead.

what flags do you use? what gn args do you have?

daescalona, 2020-11-16:
Hello,

Thse are the flags I am using in my recipe to build and launch Chromium:

`PACKAGECONFIG_append = " use-egl impl-side-painting proprietary-codecs"

CHROMIUM_EXTRA_ARGS_append = " \
        --enable-gpu \
        --enable-gpu-memory-buffer-video-frames \
        --enable-accelerated-2d-canvas \
        --enable-accelerated-vpx-decode \
        --enable-native-gpu-memory-buffers \
        --enable-zero-copy \
        --gpu-no-context-lost \
        --user-data-dir=\/home\/root\/.chromium\/ \
        --disable-infobars \
        --disable-session-crashed-bubble \
        --ignore-certificate-errors \
        --ignore-gpu-blacklist \
        --no-sandbox \
        --window-position=0,0 \
        --no-first-run \
        --in-process-gpu \
"`

Kind regards.

yifan19, 2021-04-21:
just note here that we have been using `--disable-gpu` until someone complains about performance

MonsieurHBA, 2022-03-02:
Is does configuration work for working chromium in Full screen when booting any image ?