OSSystems/meta-browser #427: Issue on drop-down list on iMX8M-Mini

ChrisGRSEB, 2020-12-22:
When I use Chromium 83.0.4103.116 on iMX8M-Mini (NXP) with BSP 5.4.47-2.2.0 and with the command : chromium --no-sandbox "URL", I have a freeze when I click on a drop-down list in order to select an item. For example, the drop-down list on this website : http://www.lehtml.com/htmlplus/fmliste.htm
Is it an issue on this chromium version ?
Is there a patch to solve this issue ?

Best regards,
Christophe

msisov, 2021-01-04:
Hi. The development of Wayland support is still going. I'd suggest to use the latest available Chromium version rather than searching for a patch to backport.

ChrisGRSEB, 2021-01-04:
Thanks @msisov for your reply. 
This bug is already known ? 
Do you know from which version of chromium this bug is fixed ?

rakuco, 2021-01-12:
It may be #423, but we'd need more information, so right now it's hard to tell if the bug is already known or not. What @msisov said still applies though: especially if you're using the chromium-ozone-wayland recipe, moving to a more recent release helps a lot, as there's a lot of development between each release and Chromium 83 is already old by Chromium's standards.

ekapllaj, 2021-06-04:
Hi, I'm working on i.MX8M Mini platform, and I also encountered a lot of issues. This is a problem with all kind of new windows created on chromium. It happens on dropdown menus, but also if you click on the three dots on the top right corner, or ajso just by right clicking and moving on the new created window, when you click outside it, chrome freezes.

The only solution I found so far is to disable this patch on the meta-imx layer (commented out):
`#GN_ARGS_append_mx8 = " use_system_minigbm=false use_wayland_gbm=false"`

With this patch, now it behaves as expected. From what I know, imx8 should support GBM. I don't know why they are disabling it for chromium. This is the commit on meta-imx (26aaeafebdc0504ca8389f7a7b04ce3595bd13c4)

> chromium-ozone-wayland: Drop wayland_gbm support for i.MX 8/8M [YOCIMX-5113] 
> This is a workaround solution for i.MX 8MQ DCSS mode

jwzl, 2021-07-16:
Me too @ekapllaj

ekapllaj, 2021-07-16:
@jwzl I tested some days ago the latest freescale BSP (Hardknott) and the overall stability was improved. They are still disabling the GBM for imx8 but now they are using "--disable-gpu-compositing" as a command line parameter. With this parameter chrome is behaving better, but the WebGL is software rendered now!
Also the full screen video on youtube works great with this parameter (without this parameter it freezes as soon as we click on full screen).
I'm now using chromium 89 with the gatesgarth BSP.

msisov, 2021-07-19:
The issues with frozen browser has been fixed by the following patch - https://github.com/OSSystems/meta-browser/blob/master/meta-chromium/recipes-browser/chromium/files/0001-Reland-ozone-wayland-watch-fd-on-a-dedicated-thread.patch

You on longer need to disable gpu compositing.

closed 2021-07-19 by msisov