OSSystems/meta-browser #445: chromium: Native builds broken on CentOS 7

r1mikey, 2021-01-25:
Just an informational note.

The native component of the Chromium build is broken on CentOS 7 (and possibly other older kernels) due to a missing `sys/random.h`.  I first noticed this with the 87.x update.  We work around this locally with the patch below, which simply includes `linux/random.h` instead.

I suspect that the best approach might be to simply document in the README that CentOS older than version 8 is not supported as a build host (given that one would need at least the `devtoolset-7` package set on the older OS anyway).

```
--- a/base/rand_util_posix.cc	2021-01-22 07:18:29.929351787 -0800
+++ b/base/rand_util_posix.cc	2021-01-22 07:18:36.139314167 -0800
@@ -24,7 +24,7 @@
 #if !defined(OS_IOS) && !defined(OS_NACL)
 // TODO(crbug.com/995996): Waiting for this header to appear in the iOS SDK.
 // (See below.)
-#include <sys/random.h>
+#include <linux/random.h>
 #endif

 namespace {
```

rakuco, 2021-01-25:
I think the missing header is a symptom of a larger issue -- it was added to glibc as a wrapper to the `getrandom()` syscall, which was added in the 3.17 Linux kernel. Looking at http://mirror.centos.org/centos/7/os/x86_64/Packages/, it looks like it ships the 3.10 kernel -- does the call to `sys_getrandom()` work at all? I wonder if CentOS ships an older kernel with a backported syscall.

rakuco, 2021-01-25:
Hmm, it's also possible that it just relies on [LSS](https://chromium.googlesource.com/linux-syscall-support/) to provide the syscall, which fails and then the code will fall back to reading `/dev/urandom`.

r1mikey, 2021-01-25:
> I think the missing header is a symptom of a larger issue -- it was added to glibc as a wrapper to the `getrandom()` syscall, which was added in the 3.17 Linux kernel. Looking at [...]

We're a special snowflake, in that we have a kernel team that updates the running kernel quite independently of the underlying OS, so the syscalls will definitely work in our environment.  The only reason we hit the error was due to the older headers.

rakuco, 2021-01-25:
I see. I think the upstream situation can be improved anyway. I've sent a [CL](https://chromium-review.googlesource.com/c/chromium/src/+/2644955) that may help here, let's see if it works and gets approved.

PR #448 (2021-01-27): chromium: Backport patch to stop including <sys/random.h>
This header might not exist on systems with very old glibc packages (e.g.
CentOS 7) when building some host binaries that we need to run as part of
the build. It turns out the header is not necessary at all, as Chromium uses
LSS (Linux Syscall Support) to call the syscall and falls back to reading
from /dev/urandom when that fails.

Fixes #445.
commit d9b0cbea (2021-01-27): chromium: Backport patch to stop including <sys/random.h>

This header might not exist on systems with very old glibc packages (e.g.
CentOS 7) when building some host binaries that we need to run as part of
the build. It turns out the header is not necessary at all, as Chromium uses
LSS (Linux Syscall Support) to call the syscall and falls back to reading
from /dev/urandom when that fails.

Fixes #445.
closed 2021-01-27 by rakuco