OSSystems/meta-browser #446: Open New Tab Page  many times will cause wl_surface@30.attach(nil, 0, 0)

cchsu23, 2021-01-25:
[log](https://drive.google.com/file/d/1eD3Y9jQ41YJJbHvgNhwLaTNmfDgBGNP_/view?usp=sharing)

When I use Chromium 86.0.4240.198 on my device to browse Netflix.
There are four links on the bottom. 
Just click more times, and  close.
WL_SURFACE seems to cannot attach original one. 
So We cannot control/touch.... but chromium seems to good.

cchsu23, 2021-02-05:
step:
1.usr/bin/chromium --app="http://www.netflix.com/login" 
2.Log-in to Netflix homepage, 
3.scroll the page on the bottom,  there are four links to facebook,ig, youtbue and twitter.
4. Choose any of links , it will open new windows , close it.
5. Repeat 4 more times.
6. You will find that the ori app mode Netflix cannot be controlled...

Remove app mode will not happened, it will open new many tab pages, and can switch back to first one Netflix.

msisov, 2021-02-05:
Can you please file this bug to crbug.com instead? thanks.

cchsu23, 2021-02-23:
Hi Msisov
https://bugs.chromium.org/p/chromium/issues/detail?id=1175376
I already file the bug to crbug. 
From owner , it can be reproduce using desktop linux x86.
I think that this issue is about  wayland backend.. 
Can you help to check log?

cchsu23, 2021-03-08:
Hi Msisov:
88.0.4342.150  has the same issue.  
--app mode  can not open new tab page

CHROMIUM_EXTRA_ARGS += " \
    --in-process-gpu \
    --enable-wayland-ime \
    --disable-restore-session-state \
    --disable-session-crashed-bubble \
    --high-dpi-support=1 \
    --force-device-scale-factor=1 \
    --touch-devices=enabled \
    --touch-events=enabled \
    --no-first-run \
    --noerrdialogs \
    --disable-pinch \
    --disable-quic \
    --ignore-gpu-blocklist \
    --disable-infobars \
    --disable-translate \
    --no-sandbox \
"

GN_ARGS += " \
    use_lld=false \
    use_gold=true \
    enable_widevine=true \
    ffmpeg_branding="Chrome" \
    proprietary_codecs=true \
    is_component_ffmpeg=true \
    media_use_ffmpeg=true \
"

my puppeteer launch  --app=https://www.netflix.com/login

msisov, 2021-03-08:
It seems like it requires to have a netflix account with a credit card added. Unfortunately, I don't use netflix and do not have an account. Maybe, you know some other way to reproduce the issue, don't you?

cchsu23, 2021-03-08:
Hi Msisov:
yes, 
1. Netflixlogin page -> Help Center  -> Watching Netflix  ->  How to download TV shows and movies to watch offline
2. Choose one of "Suggested Articles"  it also will open new tabe page.
When in app-mode it will open new browse page ,  even if i can close this page, but original one page, I can not control any more.

cchsu23, 2021-03-15:
Netflix login page    , English/Other Language   dropdown list also has the problem..

msisov, 2021-03-15:
> Netflix login page , English/Other Language dropdown list also has the problem..

The same problem has already been fixed in the master branch.

msisov, 2021-03-15:
> Hi Msisov:
> yes,
>
> 1. Netflixlogin page -> Help Center  -> Watching Netflix  ->  How to download TV shows and movies to watch offline
> 2. Choose one of "Suggested Articles"  it also will open new tabe [...]

This is also not reproducible with the master branch. Is it possible for you to grab a dev build of chromium from download-chromium.appspot.com and try again?

cchsu23, 2021-03-15:
> > Hi Msisov:
> > yes,
> > 
> > 1. Netflixlogin page -> Help Center  -> Watching Netflix  ->  How to download TV shows and movies to watch offline
> > 2. Choose one of "Suggested Articles"  it also will open new tabe page.
> >    When in app-mode it will open new browse page ,  even if i can close this page, but original one page, I can not control any more.
> 
> This is also not reproducible with the master branch. Is it possible for you to grab a dev build of chromium from https://download-chromium.appspot.com and try again?

Hi ,Msisov:
It can not be reproduced by win + x64.
It just can be reproduced by arm + linux + wayland.

msisov, 2021-03-15:
> It can not be reproduced by win + x64.

Correct. Because, Wayland is a Linux thing.

> It just can be reproduced by arm + linux + wayland.

Did you try the master branch?

cchsu23, 2021-03-15:
Hi Msisov
I always use meta-browser , due to less compile error I will met..

I will try get chromium repo and build  by myself...

rakuco, 2021-03-15:
@msisov can correct me, but I think if you just install a recent Chromium Dev package (the current one is at 91.0.4442.4) you should be able to test Wayland support on Linux without having to build everything from scratch.

msisov, 2021-03-15:
> @msisov can correct me, but I think if you just install a recent Chromium Dev package (the current one is at 91.0.4442.4) you should be able to test Wayland support on Linux without having to build everything from scratch.

That's correct. And that's why I left the link to the dev builds - https://github.com/OSSystems/meta-browser/issues/446#issuecomment-799147629

cchsu23, 2021-03-15:
Hi Msisov:
I notice that every time you release  will test  on raspberrypi3, right?
chromium-ozone-wayland:
gatesgarth, clang, MACHINE="raspberrypi3"

I think that you can use raspberrypi3 to try again, this machine is more similar with mime. 

Thanks
Dennis

cchsu23, 2021-03-15:
I think I will try 89.0.4389.82 first...

cchsu23, 2021-03-16:
89.0.4389.82 has the same problem

cchsu23, 2021-03-16:
CHROMIUM_EXTRA_ARGS += "
--in-process-gpu
--enable-wayland-ime
--disable-restore-session-state
--disable-session-crashed-bubble
--high-dpi-support=1
--force-device-scale-factor=1
--touch-devices=enabled
--touch-events=enabled
--no-first-run
--noerrdialogs
--disable-pinch
--disable-quic
--ignore-gpu-blocklist
--disable-infobars
--disable-translate
--no-sandbox
"

GN_ARGS += "
use_lld=false
use_gold=true
enable_widevine=true
ffmpeg_branding="Chrome"
proprietary_codecs=true
is_component_ffmpeg=true
media_use_ffmpeg=true
"
exec chromium:
/usr/bin/chromium --app=https://www.netflix.com/login 

Steps:
1.Netflixlogin page -> Help Center -> Watching Netflix -> How to download TV shows and movies to watch offline
2.Choose one of "Suggested Articles" it also will open new tabe page.
When in app-mode it will open new browse page , even if i can close this page, but original one page, I can not control any more.

100% crash

msisov, 2021-03-16:
not reproducible on desktop with both the lastest dev build and the stable Chrome version. I will try later on rpi.